wolfSSL/wolfssl #23: ecc25519 exports a length and type of keys

aburks, 2015-02-26:
ecc25519 exports a length and type of its public and secret key, is this necessary with explicit function calls like `wc_ecc25519_export_public`? Will there be other lengths and types (perhaps for Ed25519)?

Here's an example: https://github.com/wolfSSL/wolfssl/blob/master/wolfcrypt/src/ecc25519.c#L197-L199

JacobBarthelmeh, 2015-03-02:
Hi aburks, yeah that was initially the thought, that, along with identifying between potential future keys. It looks like the appendix of https://tools.ietf.org/html/draft-josefsson-tls-curve25519-06 is starting to solidify representation of the key for use in a TLS crypto implementation of curve25519. In the close future we will look at potentially moving more towards that type of format and not having a length and type attached.

JacobBarthelmeh, 2015-03-09:
Commit 7ef362d4bad348dceaa885ee63c1c646b20ba077 changed the format of the key to not have length and type attached.

closed 2015-03-09 by JacobBarthelmeh