wolfSSL/wolfssl #26: Compile crashes and burns on OS X :)

DomT4, 2015-03-06:
Version tarball: ` v3.4.0 `

Configure opts:

```
--disable-silent-rules --disable-dependency-tracking --infodir=/usr/local/Cellar/cyassl/3.4.0/share/info --mandir=/usr/local/Cellar/cyassl/3.4.0/share/man --prefix=/usr/local/Cellar/cyassl/3.4.0 --sysconfdir=/usr/local/etc --disable-bump --disable-examples --disable-fortress --disable-md5 --disable-sniffer --disable-webserver --enable-aesccm --enable-aesgcm --enable-blake2 --enable-camellia --enable-certgen --enable-certreq --enable-chacha --enable-crl --enable-crl-monitor --enable-dtls --enable-dh --enable-ecc --enable-eccencrypt --enable-ecc25519 --enable-filesystem --enable-hc128 --enable-hkdf --enable-inline --enable-keygen --enable-ocsp --enable-opensslextra --enable-poly1305 --enable-psk --enable-rabbit --enable-ripemd --enable-savesession --enable-savecert --enable-sessioncerts --enable-sha512 --enable-sni --enable-supportedcurves --enable-fastmath --enable-fasthugemath --enable-aesni
```

Compiler: ` Clang 6.0, Build 600 `

Failure:

```
strings -fwrapv -c src/tls.c  -fno-common -DPIC -o src/.libs/src_libwolfssl_la-tls.o
src/ssl.c:6973:9: error: use of undeclared identifier 'Md5'; did you mean 'MD5'?
        Md5  md5[1];
        ^~~
        MD5
./wolfssl/wolfcrypt/hmac.h:75:5: note: 'MD5' declared here
    MD5     = 0,
    ^
src/ssl.c:6973:12: error: expected ';' after expression
        Md5  md5[1];
           ^
           ;
src/ssl.c:6973:14: error: use of undeclared identifier 'md5'; did you mean 'md'?
        Md5  md5[1];
             ^~~
             md
src/ssl.c:6960:46: note: 'md' declared here
                       const WOLFSSL_EVP_MD* md, const byte* salt,
                                             ^
src/ssl.c:6983:20: error: use of undeclared identifier 'md5'; did you mean 'md'?
        wc_InitMd5(md5);
                   ^~~
                   md
src/ssl.c:6960:46: note: 'md' declared here
                       const WOLFSSL_EVP_MD* md, const byte* salt,
                                             ^
src/ssl.c:7023:30: error: use of undeclared identifier 'md5'; did you mean 'md'?
                wc_Md5Update(md5, digest, MD5_DIGEST_SIZE);
                             ^~~
                             md
src/ssl.c:6960:46: note: 'md' declared here
                       const WOLFSSL_EVP_MD* md, const byte* salt,
                                             ^
src/ssl.c:7025:26: error: use of undeclared identifier 'md5'; did you mean 'md'?
            wc_Md5Update(md5, data, sz);
                         ^~~
                         md
src/ssl.c:6960:46: note: 'md' declared here
                       const WOLFSSL_EVP_MD* md, const byte* salt,
                                             ^
src/ssl.c:7028:30: error: use of undeclared identifier 'md5'; did you mean 'md'?
                wc_Md5Update(md5, salt, EVP_SALT_SIZE);
                             ^~~
                             md
src/ssl.c:6960:46: note: 'md' declared here
                       const WOLFSSL_EVP_MD* md, const byte* salt,
                                             ^
src/ssl.c:7029:25: error: use of undeclared identifier 'md5'; did you mean 'md'?
            wc_Md5Final(md5, digest);
                        ^~~
                        md
src/ssl.c:6960:46: note: 'md' declared here
                       const WOLFSSL_EVP_MD* md, const byte* salt,
                                             ^
src/ssl.c:7032:30: error: use of undeclared identifier 'md5'; did you mean 'md'?
                wc_Md5Update(md5, digest, MD5_DIGEST_SIZE);
                             ^~~
                             md
src/ssl.c:6960:46: note: 'md' declared here
                       const WOLFSSL_EVP_MD* md, const byte* salt,
                                             ^
src/ssl.c:7033:29: error: use of undeclared identifier 'md5'; did you mean 'md'?
                wc_Md5Final(md5, digest);
                            ^~~
                            md
src/ssl.c:6960:46: note: 'md' declared here
                       const WOLFSSL_EVP_MD* md, const byte* salt,
                                             ^
src/ssl.c:7082:57: error: use of undeclared identifier 'Md5'
        typedef char md5_test[sizeof(MD5_CTX) >= sizeof(Md5) ? 1 : -1];
                                                        ^
src/ssl.c:7086:21: error: use of undeclared identifier 'Md5'
        wc_InitMd5((Md5*)md5);
                    ^
src/ssl.c:7086:25: error: expected expression
        wc_InitMd5((Md5*)md5);
                        ^
src/ssl.c:7094:23: error: use of undeclared identifier 'Md5'
        wc_Md5Update((Md5*)md5, (const byte*)input, (word32)sz);
                      ^
src/ssl.c:7094:27: error: expected expression
        wc_Md5Update((Md5*)md5, (const byte*)input, (word32)sz);
                          ^
src/ssl.c:7101:22: error: use of undeclared identifier 'Md5'
        wc_Md5Final((Md5*)md5, input);
                     ^
src/ssl.c:7101:26: error: expected expression
        wc_Md5Final((Md5*)md5, input);
                         ^
/bin/sh ./libtool  --tag=CC   --mode=compile clang -DHAVE_CONFIG_H -I.  -DBUILDING_WOLFSSL -fvisibility=hidden -fvisibility=hidden -DBUILDING_WOLFSSL -DOPENSSL_EXTRA -DWOLFSSL_DTLS -D_POSIX_THREADS  -DHAVE_THREAD_LS -DNDEBUG -D_THREAD_SAFE  -DPERSIST_SESSION_CACHE -DPERSIST_CERT_CACHE -DWOLFSSL_SNIFFER -DOPENSSL_EXTRA -DHAVE_AESGCM -DHAVE_AESCCM -DWOLFSSL_AESNI -maes -msse4 -DHAVE_POLY1305 -DHAVE_ONE_TIME_AUTH -DHAVE_CAMELLIA -DWOLFSSL_RIPEMD -DHAVE_BLAKE2 -DWOLFSSL_SHA512 -DWOLFSSL_SHA384 -DSESSION_CERTS -DWOLFSSL_KEY_GEN -DWOLFSSL_CERT_GEN -DWOLFSSL_CERT_REQ -DHAVE_HKDF -DNO_DSA -DHAVE_ECC -DTFM_ECC256 -DECC_SHAMIR -DHAVE_ECC25519 -DHAVE_ECC_ENCRYPT -DNO_MD5 -DNO_OLD_TLS -DNO_MD4 -DHAVE_HC128 -DHAVE_RABBIT -DHAVE_CHACHA -DHAVE_HASHDRBG -DHAVE_OCSP -DHAVE_OPENSSL_CMD -DHAVE_CRL -DHAVE_CRL_MONITOR -DHAVE_TLS_EXTENSIONS -DHAVE_SNI -DHAVE_TLS_EXTENSIONS -DHAVE_SUPPORTED_CURVES -DUSE_FAST_MATH -DUSE_FAST_MATH -Wall -Wno-unused -O2 -fomit-frame-pointer -funroll-loops -DTFM_SMALL_SET -DTFM_HUGE_SET -Wall -Wno-strict-aliasing -Wextra -Wunknown-pragmas -Waddress -Warray-bounds -Wbad-function-cast -Wchar-subscripts -Wcomment -Wfloat-equal -Wformat-security -Wformat=2 -Wmissing-field-initializers -Wmissing-noreturn -Wmissing-prototypes -Wnested-externs -Wpointer-arith -Wpointer-sign -Wredundant-decls -Wshadow -Wshorten-64-to-32 -Wsign-compare -Wstrict-overflow=1 -Wstrict-prototypes -Wswitch-enum -Wundef -Wunused -Wunused-result -Wunused-variable -Wwrite-strings -fwrapv -fPIE  -c -o src/src_libwolfssl_la-ocsp.lo `test -f 'src/ocsp.c' || echo './'`src/ocsp.c
17 errors generated.
make[1]: *** [src/src_libwolfssl_la-ssl.lo] Error 1
make[1]: *** Waiting for unfinished jobs....
```

DomT4, 2015-03-06:
If I remove the ` --disable-md5 ` from the configure args here, compile is successful, so perhaps there's an undeclared incompatibility between the configure options when ` --disable-md5 ` is invoked that I'm missing?

cconlon, 2015-03-12:
Thanks for the report.  MD5 hasn't been completely removed from our OpenSSL compatibility layer and pwdbased functionality.  This is also triggered by options that automatically enable the OpenSSL Compatibility layer (webserver, fortress, sniffer).

We'll add it to our list to refactor some of our OpenSSL Compat. and pwdbased code to try and eliminate the requirement of MD5 in those areas.

cconlon, 2015-03-26:
Thanks, should be fixed as of commit https://github.com/wolfSSL/wolfssl/commit/d4a3e83582f639ef26ac807004bb929ed988ae14

closed 2015-03-26 by cconlon

DomT4, 2015-03-26:
Thanks! Appreciate the fix. Death to MD5 :wink:.

Off-topic: Is v3.4.2 a stable release? I was going to update but noticed it isn't listed on the website or README as a released version.

cconlon, 2015-03-26:
v3.4.0 was the last stable release.  v3.4.2 is just a development tag.

DomT4, 2015-03-26:
Cool. Thanks. Wasn't sure whether the website hadn't been updated or whether it was pre-release. Thanks for the quick answer!