wolfSSL/wolfssl #27: Compile crashes and burns on OS X --with-fips

DomT4, 2015-03-06:
Different report from the one I just filed, apologies :smile_cat:.

Version `v3.4.0`.
Compiler: ` Clang 6.0, Build 600 `.

Configure opts:

```
./configure --disable-silent-rules --disable-dependency-tracking --infodir=/usr/local/Cellar/cyassl/3.4.0/share/info --mandir=/usr/local/Cellar/cyassl/3.4.0/share/man --prefix=/usr/local/Cellar/cyassl/3.4.0 --sysconfdir=/usr/local/etc --disable-bump --disable-examples --disable-fortress --disable-md5 --disable-sniffer --disable-webserver --enable-aesccm --enable-aesgcm --enable-blake2 --enable-camellia --enable-certgen --enable-certreq --enable-chacha --enable-crl --enable-crl-monitor --enable-dtls --enable-dh --enable-ecc --enable-eccencrypt --enable-ecc25519 --enable-filesystem --enable-hc128 --enable-hkdf --enable-inline --enable-keygen --enable-ocsp --enable-opensslextra --enable-poly1305 --enable-psk --enable-rabbit --enable-ripemd --enable-savesession --enable-savecert --enable-sessioncerts --enable-sha512 --enable-sni --enable-supportedcurves --enable-fastmath --enable-fasthugemath --enable-aesni --enable-fips
```

Failure:

```
In file included from ./cyassl/ctaocrypt/random.h:27:
./wolfssl/wolfcrypt/random.h:114:21: error: unknown type name 'OS_Seed'
int wc_GenerateSeed(OS_Seed* os, byte* seed, word32 sz);
                    ^
./wolfssl/wolfcrypt/random.h:125:29: error: unknown type name 'RNG'
WOLFSSL_API int  wc_InitRng(RNG*);
                            ^
./wolfssl/wolfcrypt/random.h:126:39: error: unknown type name 'RNG'
WOLFSSL_API int  wc_RNG_GenerateBlock(RNG*, byte*, word32 sz);
                                      ^
./wolfssl/wolfcrypt/random.h:127:38: error: unknown type name 'RNG'
WOLFSSL_API int  wc_RNG_GenerateByte(RNG*, byte*);
                                     ^
In file included from src/io.c:34:
In file included from ./wolfssl/internal.h:30:
In file included from ./wolfssl/crl.h:30:
In file included from ./wolfssl/wolfcrypt/asn.h:29:
In file included from ./wolfssl/wolfcrypt/rsa.h:31:
In file included from ./cyassl/ctaocrypt/rsa.h:30:
In file included from ./cyassl/ctaocrypt/random.h:27:
./wolfssl/wolfcrypt/random.h:114:21: error: unknown type name 'OS_Seed'
int wc_GenerateSeed(OS_Seed* os, byte* seed, word32 sz);
                    ^./wolfssl/wolfcrypt/random.h:131:32: error: unknown type name 'RNG'
    WOLFSSL_API int wc_FreeRng(RNG*);
                               ^

In file included from src/ssl.c:32:
In file included from ./wolfssl/internal.h:30:
In file included from ./wolfssl/crl.h:30:
In file included from ./wolfssl/wolfcrypt/asn.h:29:
In file included from ./wolfssl/wolfcrypt/rsa.h:31:
In file included from ./cyassl/ctaocrypt/rsa.h:30:
In file included from ./cyassl/ctaocrypt/random.h:27:
./wolfssl/wolfcrypt/random.h:114:21: error: unknown type name 'OS_Seed'
int wc_GenerateSeed(OS_Seed* os, byte* seed, word32 sz);
                    ^
./wolfssl/wolfcrypt/random.h:125:29: error: unknown type name 'RNG'
WOLFSSL_API int  wc_InitRng(RNG*);
                            ^
./wolfssl/wolfcrypt/random.h:126:39: error: unknown type name 'RNG'
WOLFSSL_API int  wc_RNG_GenerateBlock(RNG*, byte*, word32 sz);
                                      ^
./wolfssl/wolfcrypt/random.h:127:38: error: unknown type name 'RNG'
WOLFSSL_API int  wc_RNG_GenerateByte(RNG*, byte*);
                                     ^
In file included from src/internal.c:29:
In file included from ./wolfssl/internal.h:30:
In file included from ./wolfssl/crl.h:30:
In file included from ./wolfssl/wolfcrypt/asn.h:29:
In file included from ./wolfssl/wolfcrypt/rsa.h:31:
In file included from ./cyassl/ctaocrypt/rsa.h:30:
In file included from ./cyassl/ctaocrypt/random.h:27:
./wolfssl/wolfcrypt/random.h:114:21: error: unknown type name 'OS_Seed'
./wolfssl/wolfcrypt/random.h:131:32: error: unknown type name 'RNG'
int wc_GenerateSeed(OS_Seed* os, byte* seed, word32 sz);
                    ^
[... 271 lines cut ...]
                                    ^
./wolfssl/wolfcrypt/rsa.h:89:46: error: unknown type name 'RsaKey'
                              word32 outLen, RsaKey* key);
                                             ^
./wolfssl/wolfcrypt/rsa.h:90:36: error: unknown type name 'RsaKey'
WOLFSSL_API int  wc_RsaEncryptSize(RsaKey* key);
                                   ^
./wolfssl/wolfcrypt/rsa.h:100:41: error: unknown type name 'RsaKey'
WOLFSSL_API int  wc_RsaFlattenPublicKey(RsaKey*, byte*, word32*, byte*,
                                        ^
./wolfssl/wolfcrypt/rsa.h:104:35: error: unknown type name 'RsaKey'
    WOLFSSL_API int wc_MakeRsaKey(RsaKey* key, int size, long e, RNG* rng);
                                  ^
./wolfssl/wolfcrypt/rsa.h:104:66: error: unknown type name 'RNG'
    WOLFSSL_API int wc_MakeRsaKey(RsaKey* key, int size, long e, RNG* rng);
                                                                 ^
fatal error: too many errors emitted, stopping now [-ferror-limit=]
20 errors generated.
make[1]: *** [src/src_libwolfssl_la-internal.lo] Error 1
20 errors generated.
make[1]: *** [src/src_libwolfssl_la-tls.lo] Error 1
20 errors generated.
make[1]: *** [src/src_libwolfssl_la-ssl.lo] Error 1
make: *** [all] Error 2
```

toddouska, 2015-03-06:
Thanks for the report.  Currently FIPS isn't in our open source or basic commercial version.  Only FIPS licensed customers have access to the underlying fips files; fips.c and fps_test.c.  We made a choice to keep the --enable-fips autoconf switch so that FIPS users could plug into their existing system.  Maybe we should hide the switch in ./configure --help or have a better explanation that you probably don't have the required files.

closed 2015-03-06 by toddouska

DomT4, 2015-03-06:
Ah, thanks! I missed that announcement/message entirely. Throwing a warning or hiding the switch when the necessary files aren't detected would be cool.