wolfSSL/wolfssl #31: Defining USER_TIME should be all or nothing

jceven, 2015-03-09:
My expectation if I am defining USER_TIME is that I provide XTIME and XGMTIME, but there is no ability for me to override the XGMTIME macro.  It doesn't seem like USER is providing the time functions, if WOLFSSL is volunteering a gmtime function that can't be overridden.  USER_TIME should allow for definition of XGMTIME()

toddouska, 2015-03-12:
Hi, looking back over the code I remember the intention here.  Maybe the naming isn't ideal but the idea here is that the user has a time() type functionality but no gmtime() type functionality.  USER_TIME allows the user to override time() and use our provided gmtime().  NO_GMTIME may have been a better name.

It sounds like you need the ability to use the builtin gmtime() and override time().  Or the ability to override both (in which case your gmtime() override would call the builtin gmtime() ).  Do you have a preference?

jceven, 2015-03-13:
I think the ability to override both would be good!

toddouska, 2015-03-13:
I think the above commit should resolve the issue.  Let me know if your use cases requires any changes.  Thanks.

closed 2015-04-01 by toddouska