wolfSSL/wolfssl #35: ER: Support for compiling out unused ECC curves

aburks, 2015-03-16:
It'd be awesome to have clear `#define` options to remove unused ECC curves. For example, in our solution the smallest curve we're using is P-256 so we don't need P-112, P-128, P-160, P-192, and P-224, which saves us 1.5K of flash usage. Additionally, if we only needed P-256 we can save a total of 3K of flash usage! These calculations were based on removing the defines in `ecc.c`: https://github.com/wolfSSL/wolfssl/blob/master/wolfcrypt/src/ecc.c#L53-L60

aburks, 2015-03-16:
I understand that TLS doesn't allow you to specify your key size, which is why you don't expose this today, but I do think it'd be valuable for certain applications where the API client controls both sides of the TLS session.

toddouska, 2015-03-17:
Thanks for the request, I'll look into this.

toddouska, 2015-03-18:
defining ECC_USER_CURVES now allows the user to pick which curves they want with HAVE_ECCxxx, e.g., HAVE_ECC224.  Otherwise HAVE_ALL_CURVES is defined.  p256 is on by default in either case but can be turned off with NO_ECC256.

closed 2015-03-18 by toddouska