wolfSSL/wolfssl #37: Crash in wc_ecc_verify_hash with empty signature

briandw, 2015-03-17:
If wc_ecc_verify_hash is called with a signature of the correct length but all zeros will crash in integer.c at line 135.
I would expect to get an error for bad inputs.

Here is a sample that will crash
ecc_key pubKey;
wc_ecc_init(&pubKey);
RNG rng = [SSCryptoKeyStorage createRNG];
wc_ecc_make_key(&rng, KEY_LENGTH, &pubKey);
    
 UInt8 *hash = calloc(512, 1);
 UInt8 *bytes = calloc(24, 1);
 int verify = 0;
 wc_ecc_verify_hash(bytes, 24, hash, 512, &verify, &pubKey);

toddouska, 2015-03-17:
Thanks for the report.  I'll look into this today.

toddouska, 2015-03-17:
Fixed with this commit, forgot to reference issue in commit message: https://github.com/wolfSSL/wolfssl/commit/24ad168acfcbfc2fd7a97c276967c1b405d556d1

closed 2015-03-17 by toddouska