wolfSSL/wolfssl #48: Temporary files created by testsuite aren't removed

jay, 2015-03-23:
In testsuite.c the temporarly file name is [generated and then removed](https://github.com/wolfSSL/wolfssl/blob/9f2454f/testsuite/testsuite.c#L146-L155) before it is created. After the testing is finished the temporary file isn't removed. I'm not sure if this is intentional or not. If it's not it could be solved by using an atexit handler and `if(outputName) remove(outputName);` to remove it in all cases.

cconlon, 2015-03-26:
Hi jay,

Thanks! Could I get to you ping us at info@wolfssl.com, to get a contributor agreement in place?

Thanks,
Chris

cconlon, 2015-03-31:
Should be fixed as of this commit: https://github.com/wolfSSL/wolfssl/commit/69b4e982c8c102686c184f181c33eaedf31b180d

closed 2015-03-31 by cconlon