wolfSSL/wolfssl #52: ER: Support for 32-bit version of Ed25519

aburks, 2015-03-26:
It'd be great if there was a 32-bit native version of Ed25519 that was enabled using `ED25519_WORD32`. There is an optimized version available [here](https://github.com/floodyberry/ed25519-donna).

toddouska, 2015-03-26:
Thanks.  We are looking into this.

pzboyz, 2015-06-12:
Has this been worked on, a version of this using 32bit integer math would be very useful.

JacobBarthelmeh, 2015-06-12:
Hi pzboyz, with commit e461bc72b8149e6ef45457c1f2319bea91d30ca7 an option was added for low memory usage. This option also uses only types that are 32bit or smaller. It's a trade off between speed and memory size.

To configure with using it 
./configure --enable-sha512 --enable-ed25519=small

If on an embedded device the flag for settings.h is #define CURVED25519_SMALL and the fe_low_mem.c and ge_low_mem.c files are used.

Feel free to contact us with any questions or comments regarding it.

closed 2015-06-12 by JacobBarthelmeh