wolfSSL/wolfssl #53: ER: Support for 32-bit version of Curve25519

aburks, 2015-03-26:
It'd be great if there was a 32-bit native version of Curve25519 that was enabled using ECC25519_WORD32. There is an optimized version of Ed25519 available [here](https://github.com/floodyberry/ed25519-donna), which includes some optimized Curve25519 code that reuses the Ed25519 base.

toddouska, 2015-03-26:
Thanks for the request.  We are looking into this.  curve25519 proper is currently 32-bit but it uses fe_operations that are using 64-bit types.

JacobBarthelmeh, 2015-06-12:
With commit e461bc72b8149e6ef45457c1f2319bea91d30ca7 an option was added for low memory usage. This option also uses only types that are 32bit or smaller. It's a trade off between speed and memory size.

To configure with using it 
./configure --enable-curve25519=small

Feel free to contact us with any questions or comments regarding it.

closed 2015-06-12 by JacobBarthelmeh