wolfSSL/wolfssl #55: Calls to XGMTIME have an uninitialized variable

aburks, 2015-03-26:
Calls to `XGMTIME` in asn.c have an uninitialized variable when the function actually takes two arguments.

aburks, 2015-03-26:
Here's a quick fix:
```
diff --git a/wolfcrypt/src/asn.c b/wolfcrypt/src/asn.c
index 95208e4..2bda7ad 100644
--- a/wolfcrypt/src/asn.c
+++ b/wolfcrypt/src/asn.c
@@ -2400,9 +2400,9 @@ int ValidateDate(const byte* date, byte format, int dateType)
     struct tm* tmpTime;
     int    i = 0;
 
-#ifdef FREESCALE_MQX
-    struct tm  mqxTime;
-    tmpTime = &mqxTime;
+#if( defined( FREESCALE_MQX ) || defined( TIME_OVERRIDES ) )
+    struct tm  tmpTimeStorage;
+    tmpTime = &tmpTimeStorage;
 #else
     (void)tmpTime;
 #endif
@@ -5254,10 +5254,10 @@ static int SetValidity(byte* output, int daysValid)
     struct tm* tmpTime;
     struct tm  local;
 
-#ifdef FREESCALE_MQX
-    /* for use with MQX gmtime_r */
-    struct tm mqxTime;
-    tmpTime = &mqxTime;
+#if( defined( FREESCALE_MQX 0 || defined( TIME_OVERRIDES ) )
+    /* for use with gmtime_r */
+    struct tm tmpTimeStorage;
+    tmpTime = &tmpTimeStorage;
 #else
     (void)tmpTime;
 #endif
```

aburks, 2015-03-26:
This occurs with the `TIME_OVERRIDES` define set.

aburks, 2015-03-31:
Hey @cconlon, sorry to bug you but is there any chance you could integrate this into master?

cconlon, 2015-03-31:
Hi @aburks, we had added that 2nd argument to XGMTIME to handle the odd case where Freescale MQX didn't implement gmtime(), only gmtime_r() and needed the struct tm* passed in.

It seems like simply initializing tmpTime to NULL in the two spots we call XGMTIME in asn.c would fix the uninitialized error/warning.  We could then add a note to the docs for XGMTIME stating that the 2nd argument is only used in the case where FREESCALE_MQX is used.

Would that work for you use case?

Thanks,
Chris

aburks, 2015-03-31:
Hey Chris, thanks for the note. The gmtime() function requires static storage and may not be thread safe so I'd prefer that XGMTIME() actually uses the local storage. What do you think?

cconlon, 2015-04-01:
@aburks, that makes sense.  Fixed as requested, thanks!

closed 2015-04-01 by cconlon