wolfSSL/wolfssl #63: SHA-256 hashes are broken with Freescale MMCAU

aburks, 2015-04-02:
SHA-256 hashes are failing with a Freescale MMCAU system as of latest master (34700642fadfb90a530ca9c09299db52eb4753cb). I was using master from a week ago, commit b841bc66c31c496b6d478ac22f34d993f19f6b92.

aburks, 2015-04-02:
Here's a patch that fixes at least our use case, we need to verify that `XTRANSFORM()` always passes in buffers that are 64 bytes in length.

```
diff --git a/wolfcrypt/src/sha256.c b/wolfcrypt/src/sha256.c
index fd85363..d53f08c 100755
--- a/wolfcrypt/src/sha256.c
+++ b/wolfcrypt/src/sha256.c
@@ -446,7 +446,7 @@ int wc_Sha256Update(Sha256* sha256, const byte* data, word32 len)
         if (sha256->buffLen == SHA256_BLOCK_SIZE) {
             int ret;
 
-            #if defined(LITTLE_ENDIAN_ORDER)
+            #if (defined(LITTLE_ENDIAN_ORDER) && !defined(FREESCALE_MMCAU))
                 #if defined(HAVE_INTEL_AVX1) || defined(HAVE_INTEL_AVX2)
                 if(!IS_INTEL_AVX1 && !IS_INTEL_AVX2) 
                 #endif
@@ -481,7 +481,7 @@ int wc_Sha256Final(Sha256* sha256, byte* hash)
         XMEMSET(&local[sha256->buffLen], 0, SHA256_BLOCK_SIZE - sha256->buffLen);
         sha256->buffLen += SHA256_BLOCK_SIZE - sha256->buffLen;
 
-        #if defined(LITTLE_ENDIAN_ORDER) 
+        #if (defined(LITTLE_ENDIAN_ORDER) && !defined(FREESCALE_MMCAU))
             #if defined(HAVE_INTEL_AVX1) || defined(HAVE_INTEL_AVX2)
             if(!IS_INTEL_AVX1 && !IS_INTEL_AVX2)
             #endif
@@ -502,7 +502,7 @@ int wc_Sha256Final(Sha256* sha256, byte* hash)
     sha256->loLen = sha256->loLen << 3;
 
     /* store lengths */
-    #if defined(LITTLE_ENDIAN_ORDER)
+    #if (defined(LITTLE_ENDIAN_ORDER) && !defined(FREESCALE_MMCAU))
         #if defined(HAVE_INTEL_AVX1) || defined(HAVE_INTEL_AVX2)
         if(!IS_INTEL_AVX1 && !IS_INTEL_AVX2)
         #endif
```

toddouska, 2015-04-02:
Thanks for the report.  And the fix!

closed 2015-04-02 by toddouska