wolfSSL/wolfssl #65: options.h defines DEBUG or NDEBUG

jay, 2015-04-12:
Your configure outputs an options.h for the user that defines either `DEBUG` or `NDEBUG` which may conflict in the user's code if they or their build system already use those defines. I notice that for debug mode `DEBUG_WOLFSSL` is also defined in options.h. I wonder if it would be sufficient to keep that but filter out `NDEBUG`, `DEBUG`, and any other setting (`_POSIX_THREADS` maybe?) that could possibly cause a conflict for the user, since their program or library may not use the same settings as wolfSSL.

I recently included options.h in libcurl cyassl.c so the SNI functions in wolfSSL would be enabled and I could test against `HAVE_SNI`. I came up with some [kludge](https://github.com/jay/curl/commit/a6621b3) to workaround `NDEBUG`/`DEBUG` but I don't know if that's the right way to go. Even if you were to filter DEBUG/NDEBUG from options.h I will need to have a workaround for earlier versions. If you have a better suggestion please let me know. Off the top of my head all I can think of is an ac_check_funcs for the SNI functions in our configure, and no longer include options.h.

toddouska, 2015-04-13:
Thanks for the feedback.  I pushed a commit to remove (N)DEBUG from options.h.  I didn't yet remove _POSIX_THREADS because our wolfssl/test.h header uses it and some people use that header for simple testing / examples.

ac_check_funcs will work for SNI detection.

Another option, I'm not sure if you've explored yet, is to use user_settings.h to set wolfssl settings.  You could put 

#undef HAVE_SNI
#define HAVE_SNI

#undef HAVE_TLS_EXTENSIONS
#define HAVE_TLS_EXTENSIONS

Then define WOLFSSL_USER_SETTINGS while building or using wolfssl.  This works for autoconf and non autoconf type build systems.

jay, 2015-04-14:
Thanks. The user_settings option I can't implement though because users who are building libcurl likely have built wolfSSL many different ways so we can't rely on that. If their wolfSSL build doesn't have SNI that's ok and libcurl will still build.

jay, 2015-04-14:
Also, I just got a warning because of this undef:
```
#undef  FP_MAX_BITS 8192
#define FP_MAX_BITS 8192
```
It would probably be better to split the undef and def vars so you can do undef var like this:
```shell
undef=`echo $defonly | sed 's/=.*//'`
```

jay, 2015-04-16:
Can you tell me is it acceptable to not include the options.h file? This is what we did until recently and I don't believe it caused any problem. However is it correct? My concern is say some cyassl include has some structure that is formatted differently depending on whether some symbol is defined in options.h. libcurl uses one format but wolfSSL expects another, and bad things happen. Do you think that is possible?

toddouska, 2015-04-16:
Thanks for the undef with args report.  That's fixed.

options.h is not always required.  For example the WOLFSSL structure is perfectly opaque to the user.  This is true in general for the SSL layer at the WOLFSSL_CTX and WOLFSSL level.  On the other hand, the RSA structure is exposed to the user so that it may be used on the stack and if it is used directly by the user then any options that may alter RSA have to be consistent like USE_FAST_MATH and FP_MAX_BITS size.  We provide a configuration checker CheckCtcSettings() that returns TRUE if the underlying crypto library has the same settings.  In addition, if fastmath is used, CheckFastMathSettings() should be checked as well.

jay, 2015-04-17:
Thanks for the info and the undef fix. Going back to something you said earlier:

> I didn't yet remove _POSIX_THREADS because our wolfssl/test.h header uses it and some people use that header for simple testing / examples. 

What if "controversial" symbols that couldn't be removed could be ignored, for example like this in options.h:
```c
#undef  OPENSSL_EXTRA
#define OPENSSL_EXTRA

#ifndef CYASSL_OPTIONS_IGNORE_CONTROVERSIAL_SYMBOLS
#undef  _POSIX_THREADS
#define _POSIX_THREADS
#endif

#undef  HAVE_THREAD_LS
#define HAVE_THREAD_LS
```
The name might need some work.

toddouska, 2015-04-17:
Thanks for the idea.  I added that, I used the symbol WOLFSSL_OPTIONS_IGNORE_SYS for a little more brevity, though certainly not short.

jay, 2015-04-17:
Thanks, your recent changes work for us. I changed libcurl cyassl.c so it will only include the cyassl/options.h in versions where this issue is fixed. bagder/curl@f701125

toddouska, 2015-04-17:
Thank you, let us know if you run into anything else.

closed 2015-04-17 by toddouska