wolfSSL/wolfssl #121: DTLS 1.2 server implementation rejects valid ClientHello packets after HelloVerifySent

sebastinas, 2015-08-05:
Quoting from 4.2.1 of RFC 6347 (describing client_version and server_version in ClientHello resp. HelloVerifySent):

> DTLS 1.2 and 1.0 clients MUST use the version solely to indicate packet formatting (which is the same in both DTLS 1.2 and 1.0) and not as part of version negotiation.
> ...
> When responding to a HelloVerifyRequest, the client MUST use the same parameter values (version, random, session_id, cipher_suites, compression_method) as it did in the original ClientHello.

So when a client sends ClientHello with version set to 1.0 it needs to send 1.0 again in the ClientHello after HelloVerifyRequest regardless of what version the server sent. However, when using `wolfSSL_DTLSv1_2_server_method` for running a DTLS server, the second ClientHello is not accepted by the server and `wolfSSL_accept` fails with `VERSION_ERROR`.

I think this is caused by the check starting at https://github.com/wolfSSL/wolfssl/blob/master/src/internal.c#L3014. Changing https://github.com/wolfSSL/wolfssl/blob/master/src/internal.c#L3016 to `(ssl->options.acceptState == ACCEPT_BEGIN || ssl->options.acceptState == HELLO_VERIFY_SENT)` fixed that issue for me locally, but I'm not sure if there are checks elsewhere that need to be changed.

This issue was observed when trying to connect DTLS client using GnuTLS 3.3.8 to an DTLS server using wolfSSL 3.6.0.

kaleb-himes, 2015-08-11:
Hi sebastinas,

We wanted to thank you for your submission and update you on the status of this issue.
Preliminary work is in progress.

Kind Regards,

Kaleb

ejohnstown, 2015-09-04:
Sebastian:

This is fixed with commit 58a3656.

--John

closed 2015-09-04 by ejohnstown