wolfSSL/wolfssl #130: FREERTOS define does not allow XMALLOC_USER

aburks, 2015-08-27:
I cannot override the memory functions with the `XMALLOC_USER` define if `FREERTOS` is defined. The build will fail in settings.h around line 325.

kaleb-himes, 2015-09-21:
Hi Aburks,

Which version of wolfSSL were you using? Was line 325 at the time of this issue the #ifdef FREERTOS?

Regards,

Kaleb

aburks, 2015-09-21:
This was version 3.6.6 at commit 5ce39e147d681ba9deeb0172d190562eef223ffe.

aburks, 2015-09-21:
This also causes problems with `NO_WOLFSSL_MEMORY`. We removed our `XMALLOC_USER` code because these defines were added but then couldn't build because `XMALLOC` was redefined at line 177 of types.h.

aburks, 2015-09-21:
Additionally the FreeRTOS version of `XMALLOC` doesn't use the types, which causes build errors with -Wall turned on.

kaleb-himes, 2015-09-21:
Hi Aburks,

Thank you. I will try to get this fixed as soon as possible. Were you able to move forward with your project?

Regards,

Kaleb

aburks, 2015-09-21:
Thanks. We were able to move forward by forking and fixing it ourselves :disappointed:

kaleb-himes, 2015-09-21:
Hi Aburks,

I see. I apologize I did not get to this sooner for you.

I see you have contributed to our repository in the past, and you are the first to have implemented your own custom allocators when building with FREERTOS.

If you have a .patch file, could explain how you fixed the issue, or simply open a pull request on your branch, we would be happy to review them and see if it is something we can support going forward.

Kind Regards,

Kaleb

kaleb-himes, 2015-10-02:
Hi Aburks,

I was able to register my own callbacks for XMALLOC and XFREE with the changes made in the commit above. Were there any other conflicts that you observed and I am not aware of in regards to XMALLOC_USER and NO_WOLFSSL_MEMORY?

Regards,

Kaleb

kaleb-himes, 2015-10-06:
Fixed as of commit 67e08e48c51ade64c865531649297bc08daf4b7c. Feel free to open another issue if there is anything further.

Regards,

Kaleb

closed 2015-10-06 by kaleb-himes