wolfSSL/wolfssl #142: v3.6.6 Xcode Build failed with: inline assembly requires more registers than available

jakobsa, 2015-09-23:
adding "-fomit-frame-pointer -momit-leaf-frame-pointer" to other_c_flags solved this problem for me

jakobsa, 2015-10-20:
Having the same issue with v3.6.9. the flags still help me building the lib

My System: 
Xcode Version 7.0.1 (7A1001)
OSX 10.11 (15A284)
64Bit

ejohnstown, 2016-01-18:
Are you trying to build our iOS xcodeproj, did you modify it to build for OS X, or did you create a new project file? Normally when we build for OS X we do it from the command line.

ejohnstown, 2016-01-18:
I recreated this issue building for the iPhone 5. Our solution is to add `#define TFM_NO_ASM` to your settings.h `IPHONE` block, or the file user_settings.h, or to the Preprocessor Macros. There is a note in the README file in the v2.6.0 section regarding this.

closed 2016-01-19 by ejohnstown

jakobsa, 2017-02-04:
Thanks for coming back to me with your solution. While it might work, I still would like to leverage from your ASM code. That's why I go with my compiler flag solution. Also I have to apply those to 32bit builds only (64bit building works without those flags).

kaleb-himes, 2018-11-27:
UPDATE for future reference:

The solution here is not to disable assembly but rather to use the appropriate CFLAGS for exposing the necessary registers when using tfm.c inline assembly on i386 processor:

```
-O3 -fomit-frame-pointer
```

An example has been provided in this PR: https://github.com/wolfSSL/wolfssl/pull/1925