wolfSSL/wolfssl #152: ecc_test() fails with -1024

aburks, 2015-09-29:
Running `ecc_test()` fails with error code -1024 because validating the ECC key with `wc_ecc_check_key()` returns `FP_VAL`. This is the call stack that returns `FP_VAL`:

```
ecc_test(), test.c:5216
    wc_ecc_check_key(), ecc.c:2900
        ecc_check_pubkey_order(), ecc.c:2860
            ecc_check_pubkey_order(), ecc.c:1204
                ecc_map(), ecc.c:965
                    mp_invmod(), tfm.c:2092
```

We can reproduce this 100% of the time on our Kinetis K64 MCU.

This is on version 3.6.8.

cconlon, 2015-09-30:
Thanks for the report, I'm going to try and reproduce this on our end.  In the meantime, can you let me know what settings you were compiling wolfSSL 3.6.8 with at the time?

Thanks,
Chris

aburks, 2015-09-30:
Here's our #defines:

```C
#define FREERTOS
#define WOLFSSL_USER_IO
#define WOLFSSL_GENERAL_ALIGNMENT 4
#define WOLFSSL_SMALL_STACK
#define TFM_TIMING_RESISTANT
#define RSA_LOW_MEM
#define HAVE_ECC
#define HAVE_AESGCM
#define GCM_SMALL // GCM_TABLE || GCM_WORD32 || GCM_SMALL
#define HAVE_CURVE25519
#define HAVE_HKDF
#define WOLFSSL_BASE64_ENCODE
#define ECC_SHAMIR
#define USE_SLOW_SHA2
#define HAVE_HASHDRBG
#define HAVE_CHACHA
#define HAVE_POLY1305
#define HAVE_ONE_TIME_AUTH
#define WOLFSSL_HAVE_MIN
#define CURVED25519_SMALL

// Enable TLS Extensions and the Supported Curves extension
#define HAVE_TLS_EXTENSIONS
#define HAVE_SUPPORTED_CURVES

/* This needs to be max key size * 2 ECC256 = 512, RSA2048 = 4096 */
#if ENABLE_SSL
#define FP_MAX_BITS              4096
#else
#define FP_MAX_BITS              512
#endif
#define ALT_ECC_SIZE
#define FP_MAX_BITS_ECC          512
#define FP_LUT                   4
#define FREESCALE_MMCAU          1
#define USE_FAST_MATH /* Big stack consumer */
#define WOLFSSL_SHA512
#define HAVE_SHA512
#define HAVE_ERRNO_H
#define HAVE_LWIP_NATIVE
#define SMALL_SESSION_CACHE

/* Override time() for WolfSSL and provide our time function */
#define TIME_OVERRIDES
#define HAVE_TIME_T_TYPE
#define HAVE_TM_TYPE
#define USER_TICKS
#include "custom_time.h"

// Test-specific defines
#define BENCH_EMBEDDED
#define USE_CERT_BUFFERS_2048

#if 0
/* Debugging */
#define fprintf(file, format, ...)   printf(format, ##__VA_ARGS__)
#define DEBUG_WOLFSSL
#endif

/* remove features */
#define NO_DEV_RANDOM
#define NO_FILESYSTEM
#define NO_MAIN_DRIVER
#define NO_WOLFSSL_MEMORY
#define NO_MD4
#define NO_RABBIT
#define NO_HC128
#define NO_DSA
#define NO_PWDBASED
#define NO_PSK
#define NO_64BIT
#define NO_WOLFSSL_SERVER
#define NO_OLD_TLS
#define ECC_USER_CURVES // Disables P-112, P-128, P-160, P-192, P-224, P-384, P-521 but leaves P-256 enabled
#define NO_DES3
#define NO_MD5
#define NO_RC4
#define NO_DH
#define NO_SHA
#define CUSTOM_RAND_GENERATE  _wolfSSL_rand_gen

```

cconlon, 2015-10-02:
Are you able to show me what _wolfSSL_rand_gen() looks like?  Is it getting consistent random entropy?

manna422, 2015-11-25:
Our implementation of rand_gen uses a hardware RNG accelerator. In order to generate our seed, the least significant bit of each of the output register of the RNG is gathered and shifted by one, 32 times to produce 32 bits of entropy.

```
uint32_t _wolfSSL_rand_gen(void)
{
   uint32_t result;
   taskENTER_CRITICAL();
   // Generate 32-bits of entropy by masking the LSB of during multiple reads of a new pseudo random number number from the RNGA
   uint8_t i;
   result = 0;
   for (i = 0; i < 32; i++)
   {
      // Wait for new random number from RNGA to become available (occurs every 256 clock cycles)
      while((RNG_SR & RNG_SR_OREG_LVL(0xF)) == 0) {};

      result <<= 1;
      result |= RNG_OR & 0x0001;
   }
   taskEXIT_CRITICAL();
   return result;
}
```

I'm following up on behalf of @aburks

dgarske, 2015-12-08:
Thank you Andrew and Nicholas for the report.

We were able to locate the reason for the “ecc_test() error -1024”.

To test and reproduce this we setup a FRDM-K64F board with your user_settings. Using the latest code we were unable to reproduce, but rolling back to older versions prior to August 27th we were able to reproduce. 

We narrowed it down to the commit “August 27, 2015 - 5e26a5c (fix alt_ecc_size exptmod with negative numbers)” that resolves it. We recommend updating to a newer version of WolfSSL (at least v3.7.1).

If you decide to update to the latest v3.7.3 there is a new CUSTOM_RAND_TYPE feature you might want to consider using, which optimizes the return value on CUSTOM_RAND_GENERATE. Currently only one byte of the returned value is being used. If you add “#define CUSTOM_RAND_TYPE  word32” in your user_settings.h it will enable use of all 32-bits of returned RNG data.

Let us know if you have any further issues or questions.

closed 2015-12-08 by dgarske

aburks, 2015-12-10:
Thanks a lot for digging into this, @dgarske! We're in the process of upgrading our wolfSSL version to v3.7.1 and will let you know how it goes.