wolfSSL/wolfssl #156: WOLFSSL_CBIO_ERR_WANT_READ and WOLFSSL_CBIO_ERR_WANT_WRITE are the same number

aburks, 2015-10-06:
`WOLFSSL_CBIO_ERR_WANT_WRITE` and `WOLFSSL_CBIO_ERR_WANT_READ` are the same number, -2. This makes it difficult to determine what error was returned.

cconlon, 2015-10-07:
Hi,

We chose to leave these values the same since either wolfSSL (or an application that has registered its own I/O callbacks) already knows if it's inside the Send or Receive callback.  Once the callback returns one of these values, wolfSSL handles converting this into the correct error to return to the user.

At the application level, if wolfSSL_read()/write()/connect()/accept() returns SSL_FATAL_ERROR, the user should call ``wolfSSL_get_error()``, which will return either SSL_ERROR_WANT_READ (2) or SSL_ERROR_WANT_WRITE (3).  These are different values, allowing the user to distinguish between the two.

aburks, 2015-10-07:
Thanks for the explanation: I was seeing the expected behavior but wanted to file an issue just to be sure.

closed 2015-10-07 by aburks