wolfSSL/wolfssl #163: DTLS handshake regression

JonasNorling, 2015-10-14:
Commit 7e6b3a8 "Added more bounds checking when saving a DTLS message fragment" seems to have introduced a regression. It makes CyaSSL behave differently (worse) when running in the DTLS client role and handshake packets are dropped.

I think that the condition "fragOffset < msg->sz" should be removed so that messages without payload are allowed. The server_hello_done message is such a message.

I've tested running with the change below, and the handshake seems to be much happier.

```patch
--- cyassl.orig/src/internal.c
+++ cyassl/src/internal.c
@@ -2266,7 +2266,7 @@ void DtlsMsgSet(DtlsMsg* msg, word32 seq
                                               word32 fragOffset, word32 fragSz)
 {
     if (msg != NULL && data != NULL && msg->fragSz <= msg->sz &&
-                     fragOffset < msg->sz && (fragOffset + fragSz) <= msg->sz) {
+                     (fragOffset + fragSz) <= msg->sz) {
 
         msg->seq = seq;
         msg->type = type;
```

ejohnstown, 2015-10-27:
Thank you for the bug report. This was fixed in a slightly different manner a while ago. Your bringing this up again gave me the chance to address the fix again and your idea is a simpler fix than I had. See commit 2569cd2.

closed 2015-10-27 by ejohnstown

JonasNorling, 2015-10-28:
Thank you John.