wolfSSL/wolfssl #175: Build option to not force -j$(cores+1) from configure?

jrmithdobbs, 2015-11-03:
This would be really handy for when you're calling make of wolf from within make in another project with newer gmakes. Currently the forced setting of -j ignores the concurrency server stuff so doesn't comply with the caller's original -j option causing excessive CS in constrained build environments.

I'm hoping there's an ENV var i can set on configure that's already there that I'm missing?

kaleb-himes, 2015-11-04:
Hi jrmithdobbs,

We have the configure option to do just this, see the following results for number of jobs with the specified configure options:

> kalebs-MBP:clean_wolfssl khimes$ ```./configure``` | grep "job" 
> configure: added jobserver support to make for 9 jobs
> kalebs-MBP:clean_wolfssl khimes$ ```./configure --disable-jobserver``` | grep "job"

jrmithdobbs, 2015-11-05:
Thanks, missed it in the configure help output.

closed 2015-11-05 by jrmithdobbs