wolfSSL/wolfssl #204: Can't configure example code to use both PSK and certs.

przem312, 2015-12-02:
Hello.
In my use case, I need to configure my server to offer both PSK and certificate-based ECC cipher suites at the same time. Additionally, when client offers a certificate, the exchanged certificates must be verified on both sides (both on client and server). To achieve this, I've set following flags:
wolfSSL_CTX_set_verify( CTX_Client, 
                      SSL_VERIFY_PEER | SSL_VERIFY_FAIL_IF_NO_PEER_CERT, NULL );
        wolfSSL_CTX_set_verify( CTX_Server,
                      SSL_VERIFY_PEER | SSL_VERIFY_FAIL_IF_NO_PEER_CERT, NULL );

But when I have such configuration enabled, I cannot connect to server by using PSK ciphersuites, because wolfSSL seems to ignore the fact, that I am using PSK and expects peer to send him a certificate and as it is not sent, fails because of "SSL_VERIFY_FAIL_IF_NO_PEER_CERT". 

Is there any way to set this flag only when server selects the certificate based suite, and ignore them when PSK is chosen?

I've tried to solve the issue this way (internal.c - line 13999):           
               
                if (ssl->options.verifyPeer && ssl->options.failNoCert)

                               if (!ssl->options.usingPSK_cipher &&!ssl->options.havePeerCert) {
                                   WOLFSSL_MSG("client didn't present peer cert");
                                   return NO_PEER_CERT;
                               }

kaleb-himes, 2015-12-08:
Hi przem312,

Have you configured wolfSSL with: ```./configure --enable-psk --enable-ecc```?
Have you seen how our example server (<wolfssl-root>/examples/server/server.c) sets the psk server callback if NO_PSK is not defined?

```
633 #ifndef NO_PSK                                                                  
634         SSL_CTX_set_psk_server_callback(ctx, my_psk_server_cb);                 
...
```

An example of ```my_psk_server_cb``` can be found in: <wolfssl-root>/wolfssl/test.h

You can then enable PSK cipher suites on your server with a call to SSL_CTX_set_cipher_list:

A list of the available psk ciphers can be found in <wolfssl-root>/src/internal.c in the ```static const char* cipher_names[]```. You would need to use the string version of the cipher and pass a list of colon delimited names for example:

```
const char* cipherList = "DHE-PSK-AES256-GCM-SHA384:PSK-AES256-GCM-SHA384";
if (SSL_CTX_set_cipher_list(ctx, cipherList ) != SSL_SUCCESS)
                 err_sys("server can't set cipher list 2");
```

For usage of these two together please reference our example server again located in: <wolfssl-root>/examples/server/server.c

przem312, 2015-12-09:
Kaleb,
I've seen that, but in the very same file you have the following:

       /* if not using PSK, verify peer with certs */
    if (doCliCertCheck && usePsk == 0 && useAnon == 0) {
        SSL_CTX_set_verify(ctx, SSL_VERIFY_PEER |
                                SSL_VERIFY_FAIL_IF_NO_PEER_CERT,0);
        if (SSL_CTX_load_verify_locations(ctx, verifyCert, 0) != SSL_SUCCESS)
            err_sys("can't load ca file, Please run from wolfSSL home dir");
    }

As my use case forces me to trust the peer only if complete handshake is done (both in server context as in client context), I want to be sure that when the certificate suite is negotiated, I have the certificate exchange algorithm required. But if client hello's me only with PSK available, I don't want to check his certificate, but trade the identity and start using our key.

So basically, I'd either need to set the  

    SSL_VERIFY_PEER | SSL_VERIFY_FAIL_IF_NO_PEER_CERT,

Only both client and server agreed on certificated ciphersuite, or ignore these when PSK is negotiated (but it would have to be connection-specific, not application specific)

kaleb-himes, 2015-12-15:
Hi przem312,

So sorry it took me this long to get back to you. We had our annual team meeting last week in Seattle and I'm just getting caught up on things again.

I have some thoughts on this. As I get time this week I will be looking into it and I will get back to you as soon as possible.

Best Regards,

Kaleb

kaleb-himes, 2015-12-22:
Hi przem312,

After talking this over with other experts here at wolfSSL we have determined that what you are attempting may be possible but is not currently supported. Our suggestion would be to have two different contexts on using two separate ports. Inform all your client applications that one port is psk and one port is using certificate verification. If the client app wants to use PSK connect to PSK port else connect to certificate verification port.

We do not have plans at this time to support the run-time switching functionality between doing cert-auth, kinda sorta fail, and retry with psk. It may be possible through a consulting agreement but at this time our Engineers are all tasked out.

To discuss this in more detail would you mind opening a case in our support forums by sending an email with your desired feature request to support@wolfssl.com

Thank you so much for your question and we look forward to hearing from you via our support channel.

Best Regards,

Kaleb

closed 2015-12-22 by kaleb-himes