wolfSSL/wolfssl #263: HAVE_SECRET_CALLBACK appears to break HAVE_SESSION_TICKET

jrmithdobbs, 2016-01-18:
With code that functions using ```HAVE_SESSION_TICKET``` correctly turning on ```HAVE_SECRET_CALLBACK``` in the build and adding said callbacks to the code seems to elicit ServerHellos that, for instance openssl 1.0.1c and 1.0.2d see as invalid and abort before completing handshake with ```-tls1_2``` adding ```-no_ticket``` to the openssl s_client option allows it to continue correctly but clients eliciting support for session tickets fail hard.

In this particular usage the secret callback is being used for inspection purposes only.

Removing the HAVE_SECRET_CALLBACK and the callback setup code for it and the session ticket code begins functioning again without alteration.

jrmithdobbs, 2016-01-18:
On further inspection it only breaks if both are built and tickets are turned on and then a callback is set with ```wolfSSL_set_session_secret_cb()``` ... if it is not set tickets continue operating.

Is there a more correct way to grab a copy of the TLS master secret from inside a ```HsDoneCb``` for instance, perhaps?

jrmithdobbs, 2016-01-18:
This seems to be being caused by the two lines after this comment, as commenting them out makes things work as I'd expected: https://github.com/wolfSSL/wolfssl/blob/b78fb311bb0bd6a199a7cf1823287a5c32994533/src/ssl.c#L5077

ejohnstown, 2016-01-18:
The secret callback is there to force some other secret value that what is usually generated with TLS. Specifically, it was a feature added to support EAP-FAST which uses TLS authentication for users.

To obtain the master secret from a TLS session, use the function `wolfSSL_get_keys()`. You'll need to add `--enable-opensslextra` to the configure.

jrmithdobbs, 2016-01-18:
I was wanting to get the master secret without --enable-opensslextra, basically, by grabbing the pointer to it in this callback and then using it before it was freed in the handshake done callback.

ejohnstown, 2016-01-19:
You can move `wolfSSL_get_keys()` outside the ifdef-block.

First call `wolfSSL_KeepArrays()` on the session. The arrays won't be freed when the handshake completes. Call `wolfSSL_get_keys()` to get your master secret, do what you need to with it. Finish up with calling `wolfSSL_FreeArrrays()`.

jrmithdobbs, 2016-01-19:
Actually I don't need Keep/Free since I'm accessing it during a handshake complete callback which is always called before the buffers are cleared. I do have what I was trying to do working doing this instead of using the session secret callback.

It seems odd that everything ```--enable-opensslextra``` enables is necessary to turn it on without moving that around in the code. Was ```wolfSSL_get_keys()``` added as part of that effort? It doesn't really seem to depend on anything else behind that feature gate that I can tell.

ejohnstown, 2016-01-19:
To build WPA Supplicant against wolfSSL (at the time CyaSSL), we used the OpenSSL compatibility layer to do it. There were some functions that WPA Supplicant was using that we didn't have a compatible functions for, so we added them, `wolfSSL_get_keys()` being one. There really hasn't been much call to have it outside the opensslextras build option.

closed 2016-01-20 by ejohnstown