wolfSSL/wolfssl #294: build with picotcp is broken

JunfengJia, 2016-02-05:
In file included from ./wolfssl/internal.h:28:0,
                 from src/internal.c:31:
./wolfssl/ssl.h:962:38: error: 'struct iovec' declared inside parameter list [-Werror]
                                      int iovcnt);
                                      ^
./wolfssl/ssl.h:962:38: error: its scope is only this definition or declaration, which is probably not what you want [-Werror]
In file included from ./wolfssl/openssl/ssl.h:32:0,
                 from ./wolfssl/openssl/ec.h:6,
                 from wolfcrypt/src/ecc.c:33:
./wolfssl/ssl.h:962:38: error: 'struct iovec' declared inside parameter list [-Werror]
                                      int iovcnt);
                                      ^
./wolfssl/ssl.h:962:38: error: its scope is only this definition or declaration, which is probably not what you want [-Werror]
In file included from ./wolfssl/internal.h:28:0,
                 from src/ssl.c:34:
./wolfssl/ssl.h:962:38: error: 'struct iovec' declared inside parameter list [-Werror]
                                      int iovcnt);
                                      ^
./wolfssl/ssl.h:962:38: error: its scope is only this definition or declaration, which is probably not what you want [-Werror]
In file included from ./wolfssl/internal.h:28:0,
                 from src/io.c:36:
./wolfssl/ssl.h:962:38: error: 'struct iovec' declared inside parameter list [-Werror]
                                      int iovcnt);
                                      ^
./wolfssl/ssl.h:962:38: error: its scope is only this definition or declaration, which is probably not what you want [-Werror]
src/io.c: In function 'EmbedReceive':
src/io.c:209:27: error: implicit declaration of function 'pico_recv' [-Werror=implicit-function-declaration]
     #define RECV_FUNCTION pico_recv
                           ^
src/io.c:287:18: note: in expansion of macro 'RECV_FUNCTION'
     recvd = (int)RECV_FUNCTION(sd, buf, sz, ssl->rflags);

cconlon, 2016-02-09:
Hi,

Thanks for the report.  Was there a specific version of wolfSSL and PicoTCP that you were using to see these issues?

Thanks,
Chris

JunfengJia, 2016-02-09:
wolfssl ae19b7a272d7affff0b013f78c3fb9f68e1342d3

with current master, it works, seems you guys has fix it.

closed 2016-02-09 by JunfengJia