wolfSSL/wolfssl #326: Publickey only ECC encryption

davidgfnet, 2016-02-25:
Hello,
Is there any ciphering scheme in wolfssl that enables us to encrypt something by using a publickey only (like in RSA)?
So far I've seen there is "wc_ecc_encrypt", but this function performs a full handshake and thus requires a local private key. If I just wanted to use peer's public key I cannot get away without generating a local private key, right?
Thanks!

cconlon, 2016-03-03:
Hi David,

With ECC, wolfSSL currently only supports:

1.  ECC encrypt (wc_ecc_encrypt), which takes a public and private key, creates a shared secret, then uses that shared secret as the key for a block cipher (ie: AES) which then encrypts the input.

2.  ECC sign operations, which do a hash over the input data then sign the hash.

Can you share more about the end goal you are trying to reach, or where this operation would fit in your project?

Thanks,
Chris

cconlon, 2016-03-15:
Hi David,

Did my previous response help answer your question?  Can we close this issue?

Thanks,
Chris

davidgfnet, 2016-03-16:
Oh yeah sorry! :D Thanks!

closed 2016-03-16 by davidgfnet