wolfSSL/wolfssl #327: wc_RsaPublicEncrypt returns RSA_BUFFER_E for large input buffers

davidgfnet, 2016-02-25:
Hello,
Inspecting wc_RsaPublicEncrypt I see that if the input buffer is too big for RSA to encrypt it (by comparing to the key size) it will return RSA_BUFFER_E, but that error means "Returned if the output buffer is too small to store the ciphertext" which seems inconsistent to me.
Shouldn't that case have its own error code?
Is there a function that does split the input in chunks and process them accordingly (say for buffers that exceed the key size)? In my use case I'd like to encrypt some symmetrical AES keys using RSA, but I want to be sure that my RSA key is big enough to hold them, and otherwise split them and process them separately.

Thanks a lot!

kaleb-himes, 2016-02-25:
Hi davidgfnet,

I see your point. In reviewing our error comments on RSA_BUFFER_E in wolfssl/wolfcrypt/error-crypt.h I see ```/* RSA buffer error, output too small or input too large */```.

Is there a discrepancy in our documentation somewhere that I could fix. Where did you find the definition for that error that specifies "Returned if the output buffer is too small to store the ciphertext"?

I would be happy to update that documentation to reflect the comment in the code.

Kind Regards,

-Kaleb

davidgfnet, 2016-02-26:
Oh I see! Thanks for the clarification. The docs in the official web (might not be up to date with the source base):
https://www.wolfssl.com/wolfSSL/Docs-wolfssl-manual-18-21-wolfcrypt-api-rsa.html
So I guess it's a simple misunderstanding.

I'll take my change to ask one more question: Whenever we encrypt something with RSA, which is the usual workflow? Say we generate some random keys (+IVs) to encrypt a big file or some communication, how do we encrypt these with RSA? More precisely, what do we do if we get RSA_BUFFER_E? Reading the source it's quite clear that we should make sure the length(keys+iv) < length(RsaKey) (minus some padding) but how can we be 100% sure it will be "short enough" to fit in a single RSA encryption? Thanks!

kaleb-himes, 2016-02-26:
Hi davidgfnet,

Our engineer that maintains the online documentation is out of the office till Monday. As soon as he returns I will let him know about this issue. He will upload the updated documentation and ping you when it's update to close out this issue.

Thank you so much for bring this to our attention.

To answer your new question:

To be 100% sure that it will be short enough you could simply do a bounds check between the buffer to be encrypted and the length of the RSA key you intend to use.
For an example of the "workflow" of RSA operation I would point you to ```wolfcrypt/test/test.c``` there is a test called ```rsa_test``` in that file that provides a nice example of the typical workflow. The only thing lacking in that example is the bounds check I suggested.

Regards,

Kaleb

kaleb-himes, 2016-02-29:
Hi davidgfnet,

Talking this over with a colleague of mine just now.

You had stated:

> Say we generate some random keys (+IVs) to encrypt a big file or some communication, how do we encrypt these with RSA?

RSA is intended for use with signatures and verification of signatures. To use RSA in the way you mentioned there would be two options.

1. Generate (for example) a 1GB length RSA key to encrypt a 1GB file (probably not a good option)
2. Chop your file up into chunks that are RSA_KEY_LENGTH long encrypt each portion. (essentially turning RSA into a block cipher) and then storing them in exact order so you can decrypt them in the future.

OR

you could opt to use AES-CBC-<128/192/256> instead as AES is already a block cipher and will easily handle encryption/decryption in the way you suggested. Other options you may want to look at are aes-crt-<128/192/256> or a stream cipher such as chacha which can be used as a block cipher.

To see how you might do this we have an alpha command line utility here: https://github.com/wolfSSL/wolfssl-examples/tree/master/wolfCLU

You can follow the workflow in:
wolfCLU/src/crypto/wolfsslEncrypt.c

Regards,

-Kaleb and Jacob

kaleb-himes, 2016-03-14:
Hi davidgfnet,

Was that last helpful in your quandary?
Is there anything else before we close this github issue?

Regards,

-Kaleb

closed 2016-03-28 by kaleb-himes