wolfSSL/wolfssl #328: Missing Check for Return value on call to socket() in 'swig/wolfssl_adds.c'

dogbert2, 2016-02-25:
Hello All,

	In reviewing code in Wolfssl-3.8.0, in directory 'swig', file
'wolfssl_adds.c', there is a call to socket() in function 'tcp_socket'
which is not checked for a return value.  If the return value is < 0,
this indicates an error, which could have an impact on the code below
it which depends on variable 'sockfd'.

The patch file below should address/correct this issue:

--- wolfssl_adds.c.orig	2016-02-25 14:05:32.121623627 -0800
+++ wolfssl_adds.c	2016-02-25 14:16:39.422434128 -0800
@@ -127,10 +127,13 @@
             host = inet_ntoa(tmp.sin_addr);
         }
         else
-            return -1;   /* no entry for host */
+            return -1;	/* no entry for host */
     }
 
     *sockfd = socket(AF_INET, SOCK_STREAM, 0);
+    if (sockfd < 0)
+	return -1;  /*	unable to bind to socket    */
+
     memset(addr, 0, sizeof(SOCKADDR_IN_T));
 
     addr->sin_family = AF_INET;

=======================================================================

kaleb-himes, 2016-02-26:
Hi dogbert2,

I see we have a contributor agreement in place for you. I will bring this to the attention of the developer that worked on swig to review your suggested changes. Thank you so much for your submission.

Regards,

-Kaleb

kaleb-himes, 2016-02-29:
Hi dogbert2,

We have determined that on some systems "socket" will not always be able to return a negative value. We always write our code with portability in mind.

The error handling in tcp_connection function (beginning on line 152) will take into account any issues with the socket on the call to "connect" on line 158. While some systems cannot return negative values on the call to socket, all systems adhere to the precedent "0 for success, non-zero for failure" on the call to ```connect```.

Not checking the return code was in fact intentional in this case.

Thank you so much for submitting this issue we look forward to your review of our code and welcome any and all suggestions. It is good for us to review each case as they are pointed out and thank you for your diligence.

Kind Regards,

- Kaleb

closed 2016-02-29 by kaleb-himes