wolfSSL/wolfssl #333: Can we have some clear reference on how to use DTLS with PSK?

ShailendraGangwar, 2016-03-03:
Hi,

I am unable to use DTLS with PSK. 
Please help me in this regard.

ejohnstown, 2016-03-03:
Which platform are you running on? If you are using the configure script to set up the wolfSSL build have you tried running our example client and server with DTLS and PSK enabled? The examples and the build tests, which use the examples, have code for doing both at the same time.

ShailendraGangwar, 2016-03-04:
I am trying to implement DTLS over CoAP using PSK. I have followed https://www.wolfssl.com/wolfSSL/Blog/Entries/2013/1/23_Using_Pre-Shared_Keys_(PSK)_with_CyaSSL.html & https://www.wolfssl.com/wolfSSL/Docs-wolfssl-manual-toc.html

I have replaced all socket write calls. But I am getting socket closed everytime. Is there anything I am missing. I am using https://github.com/robbiehanson/CocoaAsyncSocket for udp connection.

ShailendraGangwar, 2016-03-04:
Hi John,

Please find attached file for the code.
I have followed this manual (
https://www.wolfssl.com/wolfSSL/Docs-wolfssl-manual-toc.html) for changing
calls for DTLS. But when running this code it throws error "UDP Socket
closed".

This code is for iOS and will run on Xcode.

ShailendraGangwar, 2016-03-04:
result = wolfSSL_write( xWolfSSL_Object, buffer, (int)length );
This line returns -1.

ejohnstown, 2016-03-04:
Did you make a call to wolfSSL_set_fd() to set the file descriptor into the WOLFSSL object?

ShailendraGangwar, 2016-03-07:
I have added following code:

/* Initialise WolfSSL.  This must be done before any other WolfSSL functions
         are called. */
        wolfSSL_Init();

        /* Attempt to create a context that uses the TLS V1 client protocol. */
        xWolfSSL_Context = wolfSSL_CTX_new( wolfDTLSv1_client_method() );

        if( xWolfSSL_Context != NULL )
        {
            wolfSSL_CTX_set_psk_client_callback(xWolfSSL_Context, My_Psk_Client_Cb);
        }

************************
int status = bind(socket4FD, (struct sockaddr *)[interface4 bytes], (socklen_t)[interface4 length]);
			if (status == -1)
			{
				[self closeSockets];
				
				NSString *reason = @"Error in bind() function";
				err = [self errnoErrorWithReason:reason];
				
				return_from_block;
			}
            else{
                xWolfSSL_Object = wolfSSL_new( xWolfSSL_Context );

                if( xWolfSSL_Object != NULL )
                {
                    /* Associate the created WolfSSL object with the connected socket. */
                    wolfSSL_set_fd( xWolfSSL_Object, socket4FD );
                }
            }

******************
result = wolfSSL_write( xWolfSSL_Object, buffer, (int)length );

ejohnstown, 2016-05-19:
Are you using our default I/O callbacks or set the peer's address with `wolfSSL_dtls_set_peer()`? In our example client, the function `tcp_connect()` sets the peer address. Our default output callback, `EmbedSendTo()`, uses the peer address in the call to `sendto()`.

ejohnstown, 2016-06-13:
I am going to close this issue at this point. If you have further problems, feel free to open a new issue or email support@wolfssl.com.

closed 2016-06-13 by ejohnstown