wolfSSL/wolfssl #348: Automated build with quality assurance

javabeanz, 2016-03-14:
Hi,

I would like to create an automated build with Travis and quality assurance. The outcome of the build can block a branch merge if quality criteria are not met. Is that something interresting for this project ? 
I have a first setup ready.

Kind regards,

javabeanz.

kaleb-himes, 2016-03-14:
Hi javabeanz,

> I would like to create an automated build with Travis and quality assurance

Were you aware we already do a significant amount of testing on every Pull Request using Jenkins (formerly Hudson) and a very robust amount of testing every night after close of business?

> The outcome of the build can block a branch merge if quality criteria are not met

Part of our "quality criteria" involve a manual code review by senior engineer only after all Jenkins tests have passed and before any merge takes place.

> Is that something interresting for this project? I have a first setup ready.

We stand by the quality of our product and more testing is always welcome! That being said is there something you feel Travis has to offer that Jenkins is not already doing?

As a side note: If you open a pull request and you are not a member of our organization your pull request will not be automatically tested by Jenkins. There is a threat vector associated with that functionality. Two criteria must be met before a third-party pull request (PR) can be tested by our Jenkins server.

1.  a contributor agreement must be signed between any third-party and wolfSSL
2. A wolfSSL engineer must first review and manually approve your PR before it is tested

Let us know your thoughts on our continuous integration practices.

Regards,

- Kaleb

javabeanz, 2016-03-14:
Yes, I've seen a presentation by another wolfssl representative on fosdem this year. I don't see that Jenkins build though, it would be nice to be more transparent here, what is tested on Jenkins is completely hidden. Travis has integrations with lots of other online tooling, often free, that helps with raising quality of code. Look at my OWASP project with the badges : https://github.com/javabeanz/owasp-security-logging.  see also : https://github.com/integrations. I don't know what agreements you refer to,

kaleb-himes, 2016-03-14:
Hi javabeanz,

You make a valid point. Those results and what is being tested is not publicly available. I will be bringing this up in our next team meeting and we will get back to you with ideas for making the results and what is being tested more third-party accessible.

The agreements I refer to are in place as a safeguard for anyone that uses our libraries. We rigorously test and review every bit of code that goes into the wolfSSL libraries for the safety and security of our end-users. To get more details on getting set up with a wolfSSL Contributor Agreement please contact support@wolfssl.com with "ATTENTION: Contributor Agreement Request" as the Subject.

Kind Regards,

- Kaleb

kaleb-himes, 2016-04-25:
Hi javabeanz,

After much internal discussion on this topic, unfortunately we can not expose the sources of our existing tests. Our testing methods are the intellectual property of wolfSSL Inc and are not GPL licensed. Our end product is GPL licensed and available to the public. We have, and will continue to offer the most tested, secure, open-source SSL implementation.

To address your previous suggestion of using travis we openly welcome you and anyone in the community to test our source code any time. Internally we feel that Travis would be duplicating tests already performed by Jenkins Continuous Integration Server.

Kind Regards,

Kaleb Himes

closed 2016-04-25 by kaleb-himes