wolfSSL/wolfssl #360: Ignore unnecessary intermediates sent by server, try alternate chains

jay, 2016-03-23:
google.com sends this chain:
~~~
 0 s:/C=US/ST=California/L=Mountain View/O=Google Inc/CN=*.google.com
   i:/C=US/O=Google Inc/CN=Google Internet Authority G2
 1 s:/C=US/O=Google Inc/CN=Google Internet Authority G2
   i:/C=US/O=GeoTrust Inc./CN=GeoTrust Global CA
 2 s:/C=US/O=GeoTrust Inc./CN=GeoTrust Global CA
   i:/C=US/O=Equifax/OU=Equifax Secure Certificate Authority
~~~
Cert 2 `GeoTrust Global CA` is an intermediate issued by root CA `Equifax Secure Certificate Authority`. That Equifax root CA was [removed](https://bugzilla.mozilla.org/show_bug.cgi?id=1156844) from Mozilla's certificate bundle late last year, and as a result it was [removed](https://github.com/bagder/ca-bundle/commit/dfcc02c#diff-0a4e35737ec12368cde6a603d2ae942cL21) from the cURL certificate bundle, which is the same as Mozilla's but in PEM format.

The Mozilla bundle still contains `GeoTrust Global CA` but [as a root certificate](https://github.com/bagder/ca-bundle/blob/dfcc02c/ca-bundle.crt#L276-L295), not an intermediate.

GeoTrust Global CA -- the intermediate CA sent by the server:
~~~
        Serial Number: 1227750 (0x12bbe6)
        Signature Algorithm: sha1WithRSAEncryption
        Issuer: C=US, O=Equifax, OU=Equifax Secure Certificate Authority
        Validity
            Not Before: May 21 04:00:00 2002 GMT
            Not After : Aug 21 04:00:00 2018 GMT
        Subject: C=US, O=GeoTrust Inc., CN=GeoTrust Global CA
~~~

GeoTrust Global CA -- the root CA in the Mozilla certificate bundle:
~~~
        Serial Number: 144470 (0x23456)
        Signature Algorithm: sha1WithRSAEncryption
        Issuer: C=US, O=GeoTrust Inc., CN=GeoTrust Global CA
        Validity
            Not Before: May 21 04:00:00 2002 GMT
            Not After : May 21 04:00:00 2022 GMT
        Subject: C=US, O=GeoTrust Inc., CN=GeoTrust Global CA
~~~

SSL libraries such as WinSSL, NSS, OpenSSL and mbedTLS can make an alternate chain and verification will succeed, but wolfSSL does not.

~~~
curl 7.48.0-DEV (i386-pc-win32) libcurl/7.48.0-DEV WinSSL WinIDN
Protocols: dict file ftp ftps gopher http https imap imaps ldap pop3 pop3s rtsp smb smbs smtp smtps telnet tftp
Features: AsynchDNS Debug IDN Largefile SSPI Kerberos SPNEGO NTLM SSL

OK:    (WinSSL uses Windows OS certificate store which contains both root CAs)

-------------------------------------------------------------------------------

curl 7.48.0-DEV (i386-pc-win32) libcurl/7.48.0-DEV OpenSSL/1.0.2g nghttp2/1.8.0
Protocols: dict file ftp ftps gopher http https imap imaps ldap pop3 pop3s rtsp smb smbs smtp smtps telnet tftp
Features: AsynchDNS Debug Largefile NTLM SSL HTTP2

OK:    Equifax Secure Certificate Authority.crt
OK:    GeoTrust Global CA.crt

-------------------------------------------------------------------------------

curl 7.48.0-DEV (i686-w64-mingw32) libcurl/7.48.0-DEV mbedTLS/2.2.1 zlib/1.2.8 libidn/1.32 libssh2/1.7.0 nghttp2/1.8.0
Protocols: dict file ftp ftps gopher http https imap imaps ldap ldaps pop3 pop3s rtsp scp sftp smtp smtps telnet tftp
Features: AsynchDNS Debug TrackMemory IDN IPv6 Largefile SSPI Kerberos SPNEGO NTLM SSL libz HTTP2

OK:    Equifax Secure Certificate Authority.crt
OK:    GeoTrust Global CA.crt

-------------------------------------------------------------------------------

curl 7.48.0-DEV (i386-pc-win32) libcurl/7.48.0-DEV wolfSSL/3.9.0
Protocols: dict file ftp ftps gopher http https imap imaps ldap pop3 pop3s rtsp smb smbs smtp smtps telnet tftp
Features: AsynchDNS Debug Largefile NTLM SSL

OK:    Equifax Secure Certificate Authority.crt
FAIL:  GeoTrust Global CA.crt
~~~

The error in the FAIL case is `ASN_NO_SIGNER_E`:
*wolfSSL Leaving SSL_get_error, return -188*
Attached: [certs.zip](https://github.com/wolfSSL/wolfssl/files/186933/certs.zip) and [wolfSSL ASN_NO_SIGNER_E debug output.txt](https://github.com/wolfSSL/wolfssl/files/186935/wolfSSL.ASN_NO_SIGNER_E.debug.output.txt).

Also see OpenSSL issue '[Support legacy CA removal, ignore unnecessary intermediate CAs in SSL/TLS handshake by default](https://rt.openssl.org/Ticket/Display.html?id=3621&user=guest&pass=guest)'.

jay, 2016-03-24:
http2.akamai.com is another example:
~~~
 0 s:/C=US/ST=CA/L=Santa Clara/O=Akamai Technologies Inc./CN=http2.akamai.com
   i:/C=NL/L=Amsterdam/O=Verizon Enterprise Solutions/OU=Cybertrust/CN=Verizon Akamai SureServer CA G14-SHA2
 1 s:/C=NL/L=Amsterdam/O=Verizon Enterprise Solutions/OU=Cybertrust/CN=Verizon Akamai SureServer CA G14-SHA2
   i:/C=IE/O=Baltimore/OU=CyberTrust/CN=Baltimore CyberTrust Root
 2 s:/C=IE/O=Baltimore/OU=CyberTrust/CN=Baltimore CyberTrust Root
   i:/C=US/O=GTE Corporation/OU=GTE CyberTrust Solutions, Inc./CN=GTE CyberTrust Global Root
~~~
Cert 2 `Baltimore CyberTrust Root` is an intermediate issued by root CA `GTE CyberTrust Global Root`. That GTE root CA was [removed](https://bugzilla.mozilla.org/show_bug.cgi?id=1047011) from Mozilla's certificate bundle last year but the Baltimore one is [a root CA in the bundle](https://github.com/bagder/ca-bundle/blob/dfcc02c/ca-bundle.crt#L112-L131). If I use just the Baltimore root CA other SSL libraries will verify it ok (tested mbedTLS, OpenSSL) but wolfSSL fails.

cconlon, 2016-03-25:
Hi Jay,

Thanks for the detailed description, we have added this to our Feature Request list.

Our position/mindset so far has been that it's not the SSL/TLS client's job to fix/create an alternate certificate chain that is different from what the server sent.  Allowing this behavior has seemed susceptible to security issues, as has been shown in the past.  For example, last year, OpenSSL had [CVE-2015-1793](https://www.openssl.org/news/secadv/20150709.txt) which was directly related to this behavior.

Thanks,
Chris

cconlon, 2016-03-29:
Added to Feature Request list, closing.

closed 2016-03-29 by cconlon

jay, 2016-03-29:
I understand, thanks for your consideration.