wolfSSL/wolfssl #361: Can not initialize initialize WOLFSSL_CTX* xWolfSSL_Context properly

r44083, 2016-03-24:
I try to use WolfSSL with freertos on STM32F407 MCU. I add WolfSSL to the project, configure the appropriate defines, so project successfully builds, but WolfSSL doesn't works. Below is part of my code.

On initialization I try to do this:
```
WOLFSSL_CTX* xWolfSSL_Context = NULL;
WOLFSSL* xWolfSSL_Object;

wolfSSL_Init();
xWolfSSL_Context = wolfSSL_CTX_new(wolfTLSv1_client_method());

wolfSSL_SetIORecv(xWolfSSL_Context, wolfssl_EmbedReceive);
wolfSSL_SetIOSend(xWolfSSL_Context, wolfssl_EmbedSend);

if(xWolfSSL_Context != NULL)
{
    wolfSSL_CTX_load_verify_buffer(xWolfSSL_Context, sizeof_ca_cert_der_1024, sizeof_ca_cert_der_1024, SSL_FILETYPE_ASN1);
}
else
{
    send_debug_msg("WolfSSL initialization error\r\n", strlen("WolfSSL initialization error\r\n"));
}
```

In this case, condition  ```if(xWolfSSL_Context != NULL)``` is always FALSE. So I get error.

This happens because in function (in /src/ssl.c, line 156) ```WOLFSSL_CTX* wolfSSL_CTX_new(WOLFSSL_METHOD* method)``` in this condition:
```
    if (method == NULL)
        return ctx;
```
```method``` always equal 0. Because function below (in /src/tls.c, line 4558) always returns 0:
```
    WOLFSSL_METHOD* wolfTLSv1_client_method(void)
    {
        WOLFSSL_METHOD* method =
                             (WOLFSSL_METHOD*) XMALLOC(sizeof(WOLFSSL_METHOD), 0,
                                                      DYNAMIC_TYPE_METHOD);
        if (method)
            InitSSL_Method(method, MakeTLSv1());
        return method;
    }
```
Function ```InitSSL_Method(method, MakeTLSv1());``` is never called.

Also need to say that I use ```#define NO_WOLFSSL_MEMORY```, so in this case XMALLOC uses standard C malloc() function.

This is my defined settings for WolfSSL:
```
#define FREERTOS
#define WOLFSSL_USER_IO
#define NO_FILESYSTEM
#define USE_CERT_BUFFERS_1024
#define NO_WOLFSSL_MEMORY
#define LARGE_STATIC_BUFFERS
#define USE_FAST_MATH
#define TFM_TIMING_RESISTANT
#define WOLFSSL_STATIC_RSA
#define WOLFSSL_STATIC_DH
#define WOLFSSL_STATIC_PSK
#define CUSTOM_RAND_GENERATE halRandom_get_number
#define NO_WOLFSSL_SERVER
#define SIZEOF_LONG 4
#define SIZEOF_LONG_LONG 8
```

Why is this happening and what's the problem?

kaleb-himes, 2016-03-24:
Hi r44083,

How did you compile wolfssl? Are you including the .c files directly in the FreeRTOS project, or are you compiling the libraries and linking against the compiled binary?

If you are linking against a compiled binary you will need the defined settings in both the library AND the FreeRTOS project to match.

If you compiled our libraries stand-alone with the autoconf (./autogen, ./configure ...) the the file <wolfssl-root>/wolfssl/options.h should have been generated and you can use ```#include <wolfssl/options.h>``` in the FreeeRTOS project to get the same settings in FreeRTOS as when compiling the library.

If you compiled the source files directly there may be something else amiss and we'll explore further when we hear back from you.

Kind Regards,

Kaleb

r44083, 2016-03-24:
I compile the source files directly (I have only c and h files) using the GNU Tools for ARM Embedded Processors 4.8.1.

dgarske, 2016-03-24:
Hey r44083,

Looks to me like your XMALLOC is returning null. Since you have FREERTOS defined the XMALLOC maps to pvPortMalloc. Curious which heap_#.c you are using in FreeRTOS? Are you using pvPortMalloc elsewhere and confident its working? Are you running out of heap space?

FreeRTOS uses 5 variants of heap management. heap_1.c, heap_2.c, etc.. so curious which one you are using. They all handle heap differently (1 simplest, 5 most complex). Heap 3 maps to libc malloc/free.

Thanks, let me know
David Garske, wolfSSL

r44083, 2016-03-24:
Hi, dgarske!
I use heap_4.c.
Except WolfSSL, ```pvPortMalloc()``` is not used anywhere in my project.

Interesting that this part of code (in /wolfcrypt/settings.h) is inactive in my project because I don't define ```XMALLOC_USER```
[image]

Is it ok?

dgarske, 2016-03-24:
Hey r44083,

Yeah that could be your problem, since having NO_WOLFSSL_MEMORY defined means it will map to malloc/free which ends up being libc, which is not part of FreeRTOS. I'd recommend removing the NO_WOLFSSL_MEMORY define from your settings and trying that again with heap_4.c. The FreeRTOS heap 4 is a good one. Let me know how that goes.

David G.

dgarske, 2016-03-24:
Also might be a good idea to enable the "#define configUSE_MALLOC_FAILED_HOOK 1" and defining the "void vApplicationMallocFailedHook( size_t xWantedSize )" function somewhere, so you can see/debug if a pvPortMalloc fails.

r44083, 2016-03-24:
When I remove ```#define NO_WOLFSSL_MEMORY``` from settings.h, looks like everything began working.
At least the initialization is successful, and I see a stream of binary data on ssl/tls port. Also I see sending a SYN request message from MCU.

dgarske, 2016-03-24:
Excellent! You are welcome. Let us know if you have any other issues. Marking closed.

closed 2016-03-24 by dgarske

r44083, 2016-03-24:
If you know, could you please tell me names of the software for debuging SSL/TLS protocol from local PC?
Since at this point the MCU sends requests and receives responses from the server, but I have no way to check if everything is ok or not. Also I will not able to solve the problem if it will occur.
At this time I use this "Hercules utility", so it doesn't allow to watch "inside" SSL or TLS:
[image]

dgarske, 2016-03-24:
There are a couple options:
1. Wireshark (https://www.wireshark.org). This will decode the un-encrypted sever/client hello. If you need to see the encrypted data there is a way to load the private key into Wireshark (see item 5: http://support.citrix.com/article/CTX116557).

2. wolfSSL sniffer. This option can be enabled via WOLFSSL_SNIFFER.
See section 4.5 of the wolfSSL manual (https://www.wolfssl.com/documentation/wolfSSL-Manual.pdf)

ejohnstown, 2016-03-24:
Windows and Wireshark have an issue with each other when using the loopback device. Windows doesn't really have one. It handles localhost with just buffer copies. There is a replacement for WinPcap that I found that seemed to work, and a lot of people are talking about it. It is called npcap. You can get an installer from their GitHub page. https://github.com/nmap/npcap

r44083, 2016-03-25:
Thx!!
I could not even imagine that the problem be resolved so quickly.