wolfSSL/wolfssl #367: What values EmbedReceive() and EmbedSend() callbacks function should return?

r44083, 2016-03-28:
Hi, I still trying to launch WolfSSL on STM32F407. At this moment I have the situation which described below:
[image]
192.168.8.128 - is my PC with Blynk server which supports SSL and TLS.
192.168.8.138 - is my MCU with WolfSSL
As you can see, MCU doesn't send "client hello".
I think that this situation because of my EmbedReceive() and EmbedSend() callbacks.

Could you please describe mу what values should returns EmbedReceive() if nothing is received?
Is EmbedSend() callback function should returns number of bytes which was send?

When I try to return 0, I get infinity loop on the internal.c:7083 because 
```while (ssl->buffers.inputBuffer.length < size);``` is always true.

Is it correct when I return size of received buffer using ```return strlen(buf);``` in my EmbedReceive() callback?

What size of TCP tx or rx buffer should I setup when using SSL or TLS with 1024 certs and raw http data is about 150 bytes?

cconlon, 2016-03-28:
Hi,

In Wireshark, are you decoding traffic as "SSL" (Analyze -> Decode As)?

> Could you please describe mу what values should returns EmbedReceive() if nothing is received?

EmbedSend() and EmbedReceive() are wolfSSL's default I/O callbacks.  The behavior of these depend on if you are using blocking or non-blocking sockets.  If using blocking sockets, EmbedReceive() returns the number of bytes received, or an error.  If using non-blocking sockets, the behavior is the same except that if the RECV function would have blocked, WOLFSSL_CBIO_ERR_WANT_READ is returned.

> Is EmbedSend() callback function should returns number of bytes which was send?

Yes, EmbedSend() should return the number of bytes sent, or an error.  If the socket is non-blocking, and the SEND function would have blocked, it will return WOLFSSL_CBIO_ERR_WANT_WRITE.

The following forum post describing wolfSSL's I/O callbacks might be helpful:
http://www.yassl.com/forums/topic760-using-custom-send-receive-callback.html

> When I try to return 0, I get infinity loop on the internal.c:7083 because 
> while (ssl->buffers.inputBuffer.length < size); is always true.

This is because wolfSSL needs to receive data in order to progress its internal state machine.  Are you using non-blocking I/O?  If so, is your application correctly handling SSL_ERROR_WANT_READ and SSL_ERROR_WANT_WRITE return values?  (and are your callbacks correctly returning the above in the case where they would normally block with blocking I/O)?

Best Regards,
Chris

r44083, 2016-03-29:
Hi,
Of course I wrote my own callbacks and add they into header file.
I corrected my callbacks functions in accordance to your comment and they seems to works now!

I enable in Wireshark (Analyze->Decode As...->SSL port 9443). Now data exchange between server and MCU using WolfSSL looks like this (TLS 1.2):
[image]

But I faced with another problem:
```wolfSSL_connect(ssl)``` function in file ssl.c:5744 returns ```"SSL_FATAL_ERROR"``` because of:

```ProcessReply(ssl)``` function in file ssl.c:5817 returns -188 ```"ASN_NO_SIGNER_E"``` because of:

on the second iteration of the ```for(;;)``` in file internal.c:7197 function ```DoHandShakeMsg()``` in file  internal.c:7409 returns -188

```DoSertificate()``` function in file internal.c:5654 returns -188 because of:

in function ```ParseCertRelative()``` in file asn.c:5013 condition ```if (ca) ``` on line 5013 is always false, so I drop into line 5044:
```
        else {
            /* no signer */
            WOLFSSL_MSG("No CA signer to verify with");
            return ASN_NO_SIGNER_E;
        }
```
and return this error -188 "ASN_NO_SIGNER_E"

Chris, could you please describe why does this error happen?
Maybe this error because of server has incorrect cert or what?

Best Regards,

I also I want to notice that the function ```DecodeToKey()``` in file asn.c:4944 returns -151 "ASN_AFTER_DATE_E".

cconlon, 2016-03-29:
An **ASN_AFTER_DATE_E** error means that when wolfSSL was validating a certificate, the certificate's "Not Valid After" date was invalid.  If you are on an embedded device, you should check to make sure your device clock is set correctly.  By default wolfSSL uses time() to get seconds since the Unix epoch.  If you have a different time function, you can define XTIME to a function that returns the seconds since Unix epoch (see the [wolfSSL Porting Guide](https://www.wolfssl.com/wolfSSL/Docs-wolfssl-porting-guide.html)).  If your clock is set correctly, it could be that one of the certificates you are using is expired.

The **ASN_NO_SIGNER_E** error means that wolfSSL could not find a CA certificate in its trust store to correctly validate the peer certificate.  Either you did not load the correct one using `wolfSSL_CTX_load_verify_locations()`, or loading failed because of the above time issue.  Are you checking the return code of `wolfSSL_CTX_load_verify_locations()`?

r44083, 2016-03-29:
I load cert from buffer this way:
```
wolfSSL_CTX_load_verify_buffer(xWolfSSL_Context, sizeof_ca_cert_der_1024, sizeof_ca_cert_der_1024, SSL_FILETYPE_ASN1);
```
Is it right?
This function returns -140 "ASN_PARSE_E" /* ASN parsing error, invalid input */

This error occurs because of function ```ParseCert()``` in ssl.c:2209 returns -140 because of:

```ParseCertRelative()``` function in asn.c:4872 returns -140 because of:

```DecodeToKey()``` function in asn.c:4944 returns -140 because of:

```GetCertHeader()``` function in asn.c:3143 -140 because of:

In function ```GetSequence()``` in asn.c:2238 condition:
```
if (input[idx++] != (ASN_SEQUENCE | ASN_CONSTRUCTED) ||
            GetLength(input, &idx, &length, maxIdx) < 0)
```
is always false (described on the sreenshot below)

[image]

If I understand correct, cert should be in the ```xWolfSSL_Context``` when I run this function ```wolfSSL_CTX_load_verify_buffer(xWolfSSL_Context, sizeof_ca_cert_der_1024, sizeof_ca_cert_der_1024, SSL_FILETYPE_ASN1);``` ?

But how can I load cert from wolfssl/certs_test.h?

cconlon, 2016-03-29:
You are passing `sizeof_ca_cert_der_1024` in for both the 2nd and 3rd function arguments.  The 2nd argument should be `ca_cert_der_1024`, which is the actual certificate buffer.

r44083, 2016-03-30:
So, what is the difference between ```XTIME callback``` and ```LowResTimer()```?

I add to the settings.h:
```
#define USER_TIME
#define USER_TICKS
```

Than I implement ```XTIME callback``` and ```LowResTimer()``` using freeRTOS:
``` xTaskGetTickCount()/1000 ```

Also I check ``` xTaskGetTickCount()/1000 ``` - it is really count every second.
So at this moment ```XTIME callback``` and ```LowResTimer()``` are the same function.
I initialize WolfSSL at time when MCU just started, so in asn.c:3020 ```XTIME(0)``` returns 0.

But I still get error -150 "ASN_BEFORE_DATE_E":
[image]

This is the place where I 
```a``` is a localTime
```b``` is a &certTime
[image]

Could you please help me to solve this?

r44083, 2016-03-30:
Important notice:
I try to replace wolfssl code in two places:

1. asn.c:2996 ```return 0;//return DateGreaterThan(b,a);```
2. asn.c:5047 ```//return ASN_NO_SIGNER_E;```

After that I was able to establish TLS 1.2 connection and send encrypted data to the server.

So, I would be glad if you share with me the ideas how to solve this.

cconlon, 2016-03-30:
**XTIME** should return the **number of seconds since the Unix epoch**.  It sounds like you are returning seconds since device reboot, which won't allow us to verify the dates in certificates correctly.  You need to switch your XTIME define to a function that returns accurate seconds since the Unix epoch in order for the certificate validation to pass the date checks.

**LowResTimer()** only needs to return a seconds count, and **does not** need to be correlated to the Unix epoch.  What you are using for this currently is fine.

Does this make sense?

r44083, 2016-03-30:
I understand you. Thx. What do you think about my previous comment?

cconlon, 2016-03-30:
If you have the time set correctly on your device, and you have correctly implemented XTIME, you shouldn't need to make any code modifications to wolfSSL like you had in your previous comment.

r44083, 2016-04-04:
Thank you Chris.

cconlon, 2016-04-06:
Hi,

Did you get this issue resolved?

Thanks,
Chris

cconlon, 2016-04-08:
Hi,

As I haven't heard back from you, I'm going to close this issue for now.  Please feel free to reopen it or create a new issue if needed.

Thanks,
Chris

closed 2016-04-08 by cconlon