wolfSSL/wolfssl #389: Statically link wolfssl on windows

plowsec, 2016-04-19:
Hi, 

Thank your for your nice work. I got used to work with wolfssl on Linux, but now I have to statically embed wolfssl into a C++ program on Windows (platform I'm not familiar with). On Linux I could pass these options to the configure script : ./configure --enable-static --enable-opensslextra --disable-fastmat and then link against wolfssl with -lwolfssl

How do I proceed on Windows to achieve the same result ? I would like to use Code::Blocks, but I'm so desperate that I'm ready to install VS or something else if needed.

cconlon, 2016-04-20:
Hi plowsec,

I haven't used Code::Blocks myself before, but the ./configure options you listed have corresponding preprocessor defines that you can add to your project's preprocessor define list to achieve the same goal.  You could also add these defines to wolfSSL's <wolfssl/wolfcrypt/settings.h> settings file, or if you have **WOLFSSL_USER_SETTINGS** defined, to your own custom <user_settings.h> header.

"--enable-opensslextra" - Equivalent of defining **OPENSSL_EXTRA**

"--disable-fastmath" - By default when not using ./configure, fastmath is turned off.  The normal big integer library is used.

"--enable-static"  - This should be a setting in your IDE/project settings to create a static library.

We don't have an IDE project file for Code::Blocks, but you should be able to follow the steps in [Section 2.4](https://www.wolfssl.com/wolfSSL/Docs-wolfssl-manual-2-building-wolfssl.html) of the wolfSSL Manual to pull wolfSSL into an existing IDE project.

Best Regards,
Chris

plowsec, 2016-04-23:
Hi Chris,

Thank you for your answer.
It appears that Code::Blocks has a feature "Import Project from MS Visual C++" in the File menu. With that and your informations, I was able to successfully and easily compile wolfssl with the settings I needed.

Kind regards,
plowsec

closed 2016-04-23 by plowsec