wolfSSL/wolfssl #396: Remove v3.69.d tag

jcowgill, 2016-04-25:
Please remove the v3.69.d tag from the repository. The tag confuses automated tools to find the latest version of wolfSSL because it ranks 3.69 above 3.9 which is the real latest version. There is already a v3.6.9d tag which points to the same commit.

For example, this is the current (trimmed) output of uscan:
```
uscan info: Matching pattern:
   (?:(?:https://github.com)?\/wolfssl\/wolfssl\/tags)?.*/v?(\d\S*)\.tar\.gz
uscan info: Found the following matching hrefs on the web page (newest first):
   /wolfSSL/wolfssl/archive/v3.69.d.tar.gz (3.69.d) index=3.69.d-1 
   /wolfSSL/wolfssl/archive/v3.9.0.tar.gz (3.9.0) index=3.9.0-1 
   /wolfSSL/wolfssl/archive/v3.8.0.tar.gz (3.8.0) index=3.8.0-1 
   /wolfSSL/wolfssl/archive/v3.7.3.tar.gz (3.7.3) index=3.7.3-1 
   /wolfSSL/wolfssl/archive/v3.7.1.tar.gz (3.7.1) index=3.7.1-1 
   /wolfSSL/wolfssl/archive/v3.7.0.tar.gz (3.7.0) index=3.7.0-1 
   /wolfSSL/wolfssl/archive/v3.6.9d.tar.gz (3.6.9d) index=3.6.9d-1 
   /wolfSSL/wolfssl/archive/v3.6.9c.tar.gz (3.6.9c) index=3.6.9c-1 
   /wolfSSL/wolfssl/archive/v3.6.9b.tar.gz (3.6.9b) index=3.6.9b-1 
   /wolfSSL/wolfssl/archive/v3.6.9.tar.gz (3.6.9) index=3.6.9-1 
uscan info: Matching target for downloadurlmangle: https://github.com/wolfSSL/wolfssl/archive/v3.69.d.tar.gz
uscan info: Upstream URL (downloadurlmangled):
   https://github.com/wolfSSL/wolfssl/archive/v3.69.d.tar.gz
uscan info: Newest upstream tarball version selected for download (uversionmangled): 3.69.d
```

cconlon, 2016-05-09:
Hi jcowgill,

Thanks for brining this to our attention.  We will work on getting it resolved.

Thanks,
Chris

cconlon, 2016-05-10:
Hi jcowgill,

We just removed the 3.69.d tag.  Thanks again for catching this.

Thanks,
Chris

closed 2016-05-10 by cconlon