wolfSSL/wolfssl #422: Buffer certs don't match file certs

ejohnstown, 2016-05-19:
The buffer version of the test certificates in the file certs_test.h do not match the certificates found in the certs directory. It looks like the file wasn't updated the last time the certificates were updated.

ejohnstown, 2016-05-19:
1. In the certificate buffers, are the names in the wrong order? Did the name parts get out of our expected order in the cert? (Do we need to modify our name matching by the parts. We have some code to do in trapped in a pull request pending a contributor agreement.)
2. Is there a bug in our name parsing?
3. Add test cases for testing the certificate buffers in addition to the certificate files.

dgarske, 2016-06-08:
Should we try and get this into the next release before the code freeze Friday?

kaleb-himes, 2016-06-08:
I believe we should yes. I’ll get this done before code-freeze on Friday.

ejohnstown, 2016-06-21:
We should be doing tests with these buffers as well, but the major problem has been fixed. Thank you, Kaleb.

closed 2016-06-21 by ejohnstown