wolfSSL/wolfssl #475: Can't compile wolfssl with GCC 6.1.1

gj12, 2016-07-11:
Hi,

I can't compile wolfssl (master from github) with gcc 6.1.1 under linux:

```
  CC       wolfcrypt/test/test.o
wolfcrypt/src/poly1305.c: In function ‘wc_Poly1305Update’:
wolfcrypt/src/poly1305.c:531:5: error: this ‘if’ clause does not guard... [-Werror=misleading-indentation]
     if (ctx == NULL)
     ^~
wolfcrypt/src/poly1305.c:535:2: note: ...this statement, but the latter is misleadingly indented as if it is guarded by the ‘if’
  if (ctx->leftover) {
  ^~
  CC       examples/client/client.o
  CC       examples/echoclient/echoclient.o
cc1: all warnings being treated as errors
make[1]: *** [Makefile:2607: wolfcrypt/src/src_libwolfssl_la-poly1305.lo] Error 1
make[1]: *** Waiting for unfinished jobs....
make[1]: Leaving directory '/root/src/wolfssl'
make: *** [Makefile:1698: all] Error 2
```

JacobBarthelmeh, 2016-07-11:
Hi gj12 thanks for bringing this to our attention. Out of curiosity is there a reason for not using a stable release of gcc, like 6.1.0? I'll work on duplicating the issue seen and get back to you with feedback. If poly1305 is not needed in the meantime it can be disabled with "./configure --disable-poly1305" while this is looked into.
Regards,
Jacob

gj12, 2016-07-12:
> Out of curiosity is there a reason for not using a stable release of gcc, like 6.1.0?

My Linux distribution (ArchLinux) installed gcc 6.1.1 with the last pacman update. (pacman -Syu). They are always using the latest versions.

https://www.archlinux.org/packages/core/x86_64/gcc/

JacobBarthelmeh, 2016-07-12:
Hello gj12,

It should be resolved now on the master branch, pull request #477 took care of the reports from gcc 6.1.1 about poly1305.c. It was catching some hard tabs in the code versus using spaces and giving warnings about code intentions near the if statement.

Thanks again for posting the issue!
Regards,
Jacob

closed 2016-07-13 by JacobBarthelmeh