wolfSSL/wolfssl #483: Got error "revcd alert fatal error" when using with libcurl.

yihuang, 2016-07-15:
I've built wolfssl 3.9.6 with libcurl-7.49.1, both are most recent release, but I've got this error when I test a https request.

```
I/stderr  ( 3341): *   Trying 14.215.177.38...
I/stderr  ( 3341): * Connected to www.baidu.com (14.215.177.38) port 443 (#0)
I/stderr  ( 3341): wolfSSL Entering WOLFSSL_CTX_new_ex
I/stderr  ( 3341): wolfSSL Entering wolfSSL_CertManagerNew
I/stderr  ( 3341): wolfSSL Leaving WOLFSSL_CTX_new, return 0
I/stderr  ( 3341): wolfSSL Entering wolfSSL_CTX_SetMinVersion
I/stderr  ( 3341): wolfSSL Entering wolfSSL_CTX_set_verify
I/stderr  ( 3341): wolfSSL Entering SSL_new
I/stderr  ( 3341): wolfSSL Leaving SSL_new, return 0
I/stderr  ( 3341): wolfSSL Entering SSL_set_fd
I/stderr  ( 3341): wolfSSL Leaving SSL_set_fd, return 1
I/stderr  ( 3341): wolfSSL Entering SSL_connect()
I/stderr  ( 3341): growing output buffer
I/stderr  ( 3341): Shrinking output buffer
I/stderr  ( 3341): connect state: CLIENT_HELLO_SENT
I/stderr  ( 3341): Embed Receive error
I/stderr  ( 3341):     Would block
I/stderr  ( 3341): wolfSSL error occurred, error = -323
I/stderr  ( 3341): wolfSSL Entering SSL_get_error
I/stderr  ( 3341): wolfSSL Leaving SSL_get_error, return -323
E/thread  ( 3341): worker task end
E/thread  ( 3341): worker thread wait
I/stderr  ( 3341): wolfSSL Entering SSL_connect()
I/stderr  ( 3341): received record layer msg
I/stderr  ( 3341): got ALERT!
I/stderr  ( 3341): Got alert
I/stderr  ( 3341): wolfSSL error occurred, error = 40
I/stderr  ( 3341): wolfSSL error occurred, error = -313
I/stderr  ( 3341): wolfSSL Entering SSL_get_error
I/stderr  ( 3341): wolfSSL Leaving SSL_get_error, return -313
I/stderr  ( 3341): wolfSSL Entering ERR_error_string
I/stderr  ( 3341): * SSL_connect failed with error -313: revcd alert fatal error
I/stderr  ( 3341): * Closing connection 0
I/stderr  ( 3341): wolfSSL Entering SSL_shutdown()
I/stderr  ( 3341): wolfSSL Leaving SSL_shutdown(), return -1
I/stderr  ( 3341): wolfSSL Entering SSL_free
I/stderr  ( 3341): CTX ref count not 0 yet, no free
I/stderr  ( 3341): wolfSSL Leaving SSL_free, return 0
I/stderr  ( 3341): wolfSSL Entering SSL_CTX_free
I/stderr  ( 3341): CTX ref count down to 0, doing full free
I/stderr  ( 3341): wolfSSL Entering wolfSSL_CertManagerFree
I/stderr  ( 3341): wolfSSL Leaving SSL_CTX_free, return 0
```

yihuang, 2016-07-15:
PS. I was building with android ndk standalone toolchain, and the configure output is:
```
Configuration summary for wolfssl version 3.9.6

   * Installation prefix:        /opt/android-toolchain-x86/usr/local
   * System type:                pc-linux-android
   * Host CPU:                   i686
   * C Compiler:                 i686-linux-android-gcc
   * C Flags:                     -g -ggdb -O0 -Wno-pragmas -Wall -Wno-strict-aliasing -Wextra -Wunknown-pragmas --param=ssp-buffer-size=1 -Waddress -Warray-bounds -Wbad-function-cast -Wchar-subscripts -Wcomment -Wfloat-equal -Wformat-security -Wformat=2 -Wmaybe-uninitialized -Wmissing-field-initializers -Wmissing-noreturn -Wmissing-prototypes -Wnested-externs -Wnormalized=id -Woverride-init -Wpointer-arith -Wpointer-sign -Wredundant-decls -Wshadow -Wsign-compare -Wstrict-overflow=1 -Wswitch-enum -Wundef -Wunused -Wunused-result -Wunused-variable -Wwrite-strings -fwrapv -fPIE
   * C++ Compiler:               i686-linux-android-g++
   * C++ Flags:                  -g -O2
   * CPP Flags:                   -fvisibility=hidden
   * CCAS Flags:
   * LIB Flags:                   -pie -z relro -z now
   * Debug enabled:              yes
   * Warnings as failure:        no
   * make -j:                    2
   * VCS checkout:               no

   Features
   * Single threaded:            no
   * Filesystem:                 yes
   * OpenSSH Build:              no
   * OpenSSL Extra API:          no
   * Max Strength Build:         no
   * fastmath:                   no
   * sniffer:                    no
   * snifftest:                  no
   * ARC4:                       no
   * AES:                        yes
   * AES-NI:                     no
   * AES-GCM:                    yes
   * AES-CCM:                    no
   * DES3:                       yes
   * IDEA:                       no
   * Camellia:                   no
   * NULL Cipher:                no
   * MD5:                        yes
   * RIPEMD:                     no
   * SHA:                        yes
   * SHA-512:                    no
   * BLAKE2:                     no
   * CMAC:                       no
   * keygen:                     no
   * certgen:                    no
   * certreq:                    no
   * certext:                    no
   * HC-128:                     no
   * RABBIT:                     no
   * CHACHA:                     yes
   * Hash DRBG:                  yes
   * PWDBASED:                   no
   * wolfCrypt Only:             no
   * HKDF:                       no
   * MD4:                        no
   * PSK:                        no
   * Poly1305:                   yes
   * LEANPSK:                    no
   * LEANTLS:                    no
   * RSA:                        yes
   * DSA:                        no
   * DH:                         yes
   * ECC:                        no
   * CURVE25519:                 no
   * ED25519:                    no
   * FPECC:                      no
   * ECC_ENCRYPT:                no
   * ASN:                        yes
   * Anonymous cipher:           no
   * CODING:                     yes
   * MEMORY:                     yes
   * I/O POOL:                   no
   * LIGHTY:                     no
   * STUNNEL:                    no
   * ERROR_STRINGS:              yes
   * DTLS:                       no
   * Old TLS Versions:           yes
   * SSL version 3.0:            no
   * OCSP:                       no
   * OCSP Stapling:              no
   * OCSP Stapling v2:           no
   * CRL:                        no
   * CRL-MONITOR:                no
   * Persistent session cache:   no
   * Persistent cert    cache:   no
   * Atomic User Record Layer:   no
   * Public Key Callbacks:       no
   * NTRU:                       no
   * Whitewood netRandom:        no
   * Server Name Indication:     no
   * ALPN:                       no
   * Maximum Fragment Length:    no
   * Truncated HMAC:             no
   * Supported Elliptic Curves:  no
   * Session Ticket:             no
   * Renegotiation Indication:   no
   * Secure Renegotiation:       no
   * All TLS Extensions:         no
   * PKCS#7                      no
   * wolfSCEP                    no
   * Secure Remote Password      no
   * Small Stack:                no
   * valgrind unit tests:        no
   * LIBZ:                       no
   * Examples:                   yes
   * User Crypto:                no
   * Fast RSA:                   no
   * Async Crypto:               no
```

yihuang, 2016-07-15:
Build with ``--with-ecc`` solved my problem.

cconlon, 2016-07-15:
Thanks for letting us know you got the issue resolved.

closed 2016-07-15 by cconlon