wolfSSL/wolfssl #507: Python wrapper for wolfssl

hsuyh, 2016-07-28:
Hi,

I think the wolfSSL for C language has already support ECDHE_ECDSA.  
But I can't saw any python example about ECDHE_ECDSA.
I am curious about did python wrapper for wolfSSL(wolfCrypt Python) support the ECHDE_ECDSA?

Thank you!

moisesguimaraes, 2016-07-31:
Hi @hsuyh ,

ECDHE-ECDSA was not in the scope of our first **wolfcrypt python** module release.  Could you please share your interest and experience in our python module?  Meanwhile I'll talk to our other engineers to check the next milestones for **wolfcrypt python** to see if ECDHE_ECDSA is already contemplated.

[ ]'s

hsuyh, 2016-08-01:
Hi @moisesguimaraes ,

I want to implement the security mechanisms in the IoT platform, which was written in python. The security mechanisms are defined by the international IoT standard oneM2M. The standard asks to use ECDHE-ECDSA for one of the security mechanism. If wolfcrypt supports the ECDHE_ECDSA, I can use your library to implement those security mechanisms. That is what I want to do.   ECDHE_ECDSA is more secure and faster than ECDHE_RSA. If you can support this, it will be better! Thank you!

ejohnstown, 2019-02-22:
We recently put some more work into the wolfCrypt and wolfSSL Python wrappers. I am currently able to run the wolfssl-py example client and server connecting with TLSv1.2 and the cipher suite ECDHE-ECDSA-AES128-GCM-SHA256. (I did have to use the example certs out of the wolfSSL project, rather than the example certs included in the wolfssl-py project.) I am also able to negotiate with TLSv1.3 using the same ECC certificate and key.

Thank you for your interest in wolfSSL!

closed 2019-02-22 by ejohnstown