wolfSSL/wolfssl #508: Could you upload a ready-to-use Arduino version of this library to its official Library Manager?

q2dg, 2016-07-28:
This would give to this project a lot more visibility and traction, I think.
Steps to do the process are here: https://github.com/arduino/Arduino/wiki/Library-Manager-FAQ
Thanks a lot.

P.S: It would be great if this example sketch (https://github.com/wolfSSL/wolfssl/blob/190d30f336c0ff32aecbf198bd0603b42dcac46a/IDE/ARDUINO/sketches/wolfssl_client.ino ) -or others if exist- come along with the library installation inside "Examples" folder

cconlon, 2016-08-23:
Hi,

Thanks for the suggestion.  We'll put this on our list of projects to look into.

We did our Arduino port a while back, and have a script under the ./IDE/ARDUINO directory  (wolfssl-arduino.sh) that rearranges the header files to match the requirements of the Arduino IDE.

Best Regards,
Chris

haydenroche5, 2021-01-13:
Hi,

For anyone interested, this request is on our internal list of feature requests. If users are interested in us doing this, please reach out.

Thanks,

Hayden

closed 2021-01-13 by haydenroche5