wolfSSL/wolfssl #510: Zero padding problem in ParseCert(asn.c)

PentaZoo, 2016-08-01:
- in wolfssl 3.9.0 Environment

1) I saved the certificate file in the buffer.
2) I saved the CRL file in the buffer.
 - Revocation lists from CRL has an 1) certificate information.
=> I expect the alert via CRL verification (wolfSSL_CertManagerCheckCRL). 
 However, the certificate is passed through a CRL test in my situation.

I had to analyze the reasons, it found the matter of ParseCert (asn1).
The problem was in part to parse the serial.

- CRL has the following serial as the certificate revocation list 
 : 0x00 0xe8 0xd4 0xa5 0x10 0x63 (length: 6) 

But, the certificate to parse Cert has the following serial 
 : 0xe8 0xd4 0xa5 0x10 0x63 (length: 5) 

Therefore, when the CRL checking the two serial, wolfssl has determined that there is no problem.
I think the problem is that the zero padding.
00 e8 d4 a5 10 63 ==== parseCert ===> e8 d4 a5 10 63

I want you to want me suggest a solution.
Thanks.

ejohnstown, 2016-08-01:
Is there a chance you can share with me the cert and CRL you are testing with? You can send it to me at our support address, support@wolfssl.com.

PentaZoo, 2016-08-02:
> Is there a chance you can share with me the cert and CRL you are testing with? You can send it to me at our support address, support@wolfssl.com <mailto:support@wolfssl.com>.

Dear.

I attach files the cert and CRL.
but, this CRL is expired (2016.07.31-2016.08.01)
I requested new CRL files to manager, but I don’t receive yet.

- in wolfssl 3.9.0 Environment

1) I saved the certificate file in the buffer.
2) I saved the CRL file in the buffer.
 - Revocation lists from CRL has an 1) certificate information.
=> I expect the alert via CRL verification (wolfSSL_CertManagerCheckCRL). 
 However, the certificate is passed through a CRL test in my situation.

I had to analyze the reasons, it found the matter of ParseCert (asn1).
The problem was in part to parse the serial.

- CRL has the following serial as the certificate revocation list 
 : 0x00 0xe8 0xd4 0xa5 0x10 0x63 (length: 6) 

But, the certificate to parse Cert has the following serial 
 : 0xe8 0xd4 0xa5 0x10 0x63 (length: 5) 

Therefore, when the CRL checking the two serial, wolfssl has determined that there is no problem.
I think the problem is that the zero padding.
00 e8 d4 a5 10 63 ==== parseCert ===> e8 d4 a5 10 63

I want you to want me suggest a solution.
Thanks.

dgarske, 2016-08-03:
Hello PentaZoo,

I've carefully reviewed the code and confirmed there is an issue in the "GetRevoked" function which does not trim the padding. So when calling "wolfSSL_CertManagerCheckCRL" it was not matching the CRL against the certificate.

Fixed in PR #512. This will go through normal internal testing and should get into master later this week. Let me know if you see any issues when you try it.
https://github.com/wolfSSL/wolfssl/pull/512

Also, I pushed a CRL check example to:
https://github.com/wolfSSL/wolfssl-examples/pull/21

Thanks,
David Garske

closed 2016-08-03 by dgarske