wolfSSL/wolfssl #539: Compiling library on arduino IDE 1.6.11 / Ubuntu 16.04.1

AnthonyDeroche, 2016-09-03:
Hello,

I'm trying to compile the lib under Arduino IDE 1.6.11 / Ubuntu 16.04.1.

Compilation performed without errors once I have carried few steps out:
- execute script IDE/ARDUINO/wolfssl-arduino.sh (I had to modify "==" by "=" in condition)
- uncomment "#define WOLFSSL_ARDUINO" in wolfssl/wolfcrypt/settings.h
- add "#define NO_INLINE" in wolfssl/wolfcrypt/settings.h
- Implement GenerateSeed in random.c (line 1652) (I basically uncommented the commented function)

I'm wondering if I should really do all of this in order to pass compilation under Arduino IDE ?
Might these issues come from my environment?

Thanks,
Anthony

kaleb-himes, 2016-09-14:
Hi Anthony,

Thanks for your question! I'm currently looking into your case and will get back to as soon as I have an update.

- Kaleb

kaleb-himes, 2016-09-14:
Hi Anthony,

I have created a pull request here that addresses some of your concerns: https://github.com/wolfSSL/wolfssl/pull/551

> Implement GenerateSeed in random.c (line 1652) (I basically uncommented the commented function)

This is necessary as we do not provide a source of entropy with our libraries. We use /dev/urandom when available and many embedded systems have their own hardware entropy source which can be setup and accessed through the GenerateSeed API.

I did not find it necessary to define NO_INLINE when building for Intel® Galileo board (the only platform we currently have ported to in ARDUINO)

By defining INTEL_GALILEO there is no need to remove references to XTIME/GMTIME, I've added this to the README and a commented define under WOLFSSL_ARDUINO in wolfssl/wolfcrypt/settings.h to make this more intuitive.

I updated the /bin/sh script as you mentioned.

Let us know if there is anything further.

Kind Regards,
Kaleb

AnthonyDeroche, 2016-09-15:
Thanks for your fast reply.

I will try it out as soon as possible. I will keep you posted.

Regards,
Anthony

cconlon, 2016-10-10:
Hi Anthony,

Just checking in on this issue.  Were Kaleb's comments above helpful?

Thanks,
Chris

AnthonyDeroche, 2016-10-10:
I didn't try again to compile the library in my environment. I will keep you posted as soon as I work again in this project.

Thanks,
Anthony

kaleb-himes, 2017-04-06:
Hi Anthony,

Did you ever get a chance to get back to this? Should we close this issue?

Regards,

Kaleb

pasko-zh, 2017-04-13:
Hello wolfSSL!

I am posting this here, because it seems to be related (If you wish I can also open a new issue)

I am trying to compile a simple sketch with wolfcrypt (SHA 256 HMAC) for an esp8266 with the Arduino tool chain. I followed the steps [here](https://github.com/wolfSSL/wolfssl/tree/master/IDE/ARDUINO) and read the thread [here](https://www.wolfssl.com/forums/topic861-solved-getting-started-wolfcrypt-on-arduino.html).

I am using the following versions

- wolfSSL, 3.10.2-stable
- Arduino 1.8.2
- Arduino esp core 2.3.0

When I am working with Visual Studio (together with the visual micro plugin), the following code 
```c 
// This sketch is saved as wolfhmactest.ino

#include <wolfssl.h>
#include <wolfcrypt\sha256.h>
#include <wolfcrypt\hmac.h>

void hmac_sha256_test() {
	Hmac hmac_256;
	int ret;
	// "Jefe", or { 74, 101, 102, 101 }
	const char *key_1[] = { "Jefe" };
	ret = wc_HmacSetKey(&hmac_256, 2, (byte *)key_1, (word32) 4);
}

void setup() {
	Serial.begin(115200);
	delay(1000);
}

void loop() {
	hmac_sha256_test();
	delay(5000);
}

```  
compiles with the following errors:

````
Compiling 'wolfhmactest' for 'Adafruit HUZZAH ESP8266'
wolfhmactest.cpp.o*: (.text._Z16hmac_sha256_testv+0x4): undefined reference to wc_HmacSetKey

Error linking for board Adafruit HUZZAH ESP8266
Build failed for project 'wolfhmactest'
 
wolfhmactest.cpp.o*: In function hmac_sha256_test()
wolfhmactest.ino:11: undefined reference to wc_HmacSetKey
 
collect2.exe*: error: ld returned 1 exit status
````

When I try to compile with the (native) Arduino IDE, I need to change some entries in the header files first, otherwise the IDE won't find them. For instance I need to change `#include <wolfssl/wolfcrypt/types.h>` to `#include <wolfcrypt/types.h>`. When having done so, the error message of the compiler is the same as above.

Thanks for your feedback!
paško

closed 2017-04-27 by JacobBarthelmeh

apullin, 2017-05-07:
I had a follow-up on this issue, although maybe I should open another issue for it:

I am building for a more plain Arduino platform, not a Yun or Galileo or any of the linux-backed platforms, just a simple microcontroller embedded context where a WifiClient that conforms to the expected interface is provided (ESP8266 derivative).

When building, I get the errors:
```
/var/folders/dy/nsmz0p8s06b0z0_rrfvqtqz40000gn/T/arduino_build_70070/libraries/wolfssl/asn.c.o: In function `CheckCurve':
/Users/xyz/Documents/Arduino/libraries/wolfssl/asn.c:4036: undefined reference to `__ctype_ptr__'
/var/folders/dy/nsmz0p8s06b0z0_rrfvqtqz40000gn/T/arduino_build_70070/libraries/wolfssl/asn.c.o: In function `ExtractDate':
/Users/xyz/Documents/Arduino/libraries/wolfssl/asn.c:4036: undefined reference to `XTIME'
/Users/xyz/Documents/Arduino/libraries/wolfssl/asn.c:4036: undefined reference to `XGMTIME'
/var/folders/dy/nsmz0p8s06b0z0_rrfvqtqz40000gn/T/arduino_build_70070/libraries/wolfssl/asn.c.o: In function `ValidateDate':
/Users/xyz/Documents/Arduino/libraries/wolfssl/asn.c:4036: undefined reference to `XTIME'
/Users/xyz/Documents/Arduino/libraries/wolfssl/asn.c:4036: undefined reference to `XGMTIME'
collect2: error: ld returned 1 exit status
```

When I try to define even a placeholder function like below in the .ino file to get the link error to resolve (not expecting it to work properly), the problem remains.
```
extern time_t XTIME(time_t * timer){
  time_t temp;
  return temp;
}
```

Am I overlooking something for the correct procedure for how to provide XTIME and XGMTIME to the settings and configuration?

apullin, 2017-05-14:
tagging the pros so they'll see by above comment, before I open a separate issue, @kaleb-himes @cconlon

flacidsnake, 2018-04-06:
Hi,

I know this is closed, but I'm having similar problems as @AnthonyDeroche 

After doing what he posted, I'm having these compilation problems:

arduino-1.8.5\libraries\wolfssl/wolfssl/wolfcrypt/wc_port.h:469:31: error: expected unqualified-id before 'const'

     extern struct tm* XGMTIME(const time_t* timer, struct tm* tmp);

                               ^

E:\arduino-1.8.5\libraries\wolfssl/wolfssl/wolfcrypt/wc_port.h:469:31: error: expected ')' before 'const'

E:\arduino-1.8.5\libraries\wolfssl/wolfssl/wolfcrypt/wc_port.h:469:31: error: expected initializer before 'const'

exit status 1
Error compilando para la tarjeta Arduino Leonardo.

Files I think are involved are pasted below.

**_My file settings.h_**
`
/* settings.h
 *
 * Copyright (C) 2006-2017 wolfSSL Inc.
 *
 * This file is part of wolfSSL.
 *
 * wolfSSL is free software; you can redistribute it and/or modify
 * it under the terms of the GNU General Public License as published by
 * the Free Software Foundation; either version 2 of the License, or
 * (at your option) any later version.
 *
 * wolfSSL is distributed in the hope that it will be useful,
 * but WITHOUT ANY WARRANTY; without even the implied warranty of
 * MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.  See the
 * GNU General Public License for more details.
 *
 * You should have received a copy of the GNU General Public License
 * along with this program; if not, write to the Free Software
 * Foundation, Inc., 51 Franklin Street, Fifth Floor, Boston, MA 02110-1335, USA
 */

/* Place OS specific preprocessor flags, defines, includes here, will be
   included into every file because types.h includes it */
#define XTIME fnSecondsSinceEpoch
#define XGMTIME

#ifndef WOLF_CRYPT_SETTINGS_H
#define WOLF_CRYPT_SETTINGS_H

#ifdef __cplusplus
    extern "C" {
#endif

/* Uncomment next line if using IPHONE */
/* #define IPHONE */

/* Uncomment next line if using ThreadX */
/* #define THREADX */

/* Uncomment next line if using Micrium uC/OS-III */
/* #define MICRIUM */

/* Uncomment next line if using Mbed */
/* #define MBED */

/* Uncomment next line if using Microchip PIC32 ethernet starter kit */
/* #define MICROCHIP_PIC32 */
[... 2053 lines cut ...]
    extern time_t XTIME(time_t * timer);

    #ifdef STACK_TRAP
        /* for stack trap tracking, don't call os gmtime on OS X/linux,
           uses a lot of stack spce */
        extern time_t time(time_t * timer);
        #define XTIME(tl)  time((tl))
    #endif /* STACK_TRAP */

#elif defined(TIME_OVERRIDES)
    extern time_t XTIME(time_t * timer);
    extern struct tm* XGMTIME(const time_t* timer, struct tm* tmp);
#elif defined(WOLFSSL_GMTIME)
    struct tm* gmtime(const time_t* timer);
#endif
#endif /* NO_ASN_TIME */

#ifdef __cplusplus
    }  /* extern "C" */
#endif

#endif /* WOLF_CRYPT_PORT_H */

`
`