wolfSSL/wolfssl #568: DTLS pool size

steweg, 2016-09-20:
Hi,

we have found a bug with DTLS pool size. The default DTLS pool size is set to be 5, which seems to be enough for DTLS handshake messages in general. But you forget about the fragmentation. If you use certificates bigger than one message, the pool is not enough. So when one of the message in the DTLS flight get lost, the handshake never completes as during retransmission, again some message will be not sent at all.

So what we did as a quick fix was to increase the DTLS pool size to 10. In long term, this should be rather dynamic than static, which would allow to store any number of messages including fragmented ones. So rather than creating PULL request for my quick fix, I prefer to leave it up to you to find proper solution.

What is needed in order to reproduce the problem is:
* DTLS server with bigger certificates (including some certificate chain), that will cause fragmentation
* DTLS client
* virtual network between DTLS client and DTLS server with artificial packet-loss (e.g. 10%)
* run the client and server
* used version is 3.9.8

Observations from our tests:
* we have seen infinite sending of messages back and forth between client and sender, until our timeout function didn't stop the processing. This caused a lot of network traffic in quite a short time
* even after our quick fix, there seems to be another bug related to retranmissions, but we are not yet sure what is causing it. If you increase the number of clients to 50 and all of them would like to do just handshake at the same time, some of them exchange only few messages (<20), but some seem to exchange a lot more in some kind of loops (>350 messages). We will continue to work on this issue as our target is to have DTLS server able to handle 10K+ clients.

ejohnstown, 2016-09-20:
Thanks for the report. I'll look into the pool size issue soon. I look forward to your ananlysis. Could you please redirect that to our support system (mailto:support@wolfssl.com)?

ejohnstown, 2016-11-01:
This has been fixed with commit 3075269 in pull request #606. The other issue was fixed by your commit 3065bb2 in pull request #588. Thank you for the patch.

closed 2016-11-01 by ejohnstown