wolfSSL/wolfssl #610: Misleading SRP API

damv, 2016-11-02:
When calling wc_SrpGetPublic() function, if no SRP private key has been manually generated, it generates a default one.
In this case the size of the private key is equals to modulus size.
I think it is a misleading value, since in RFC https://tools.ietf.org/html/rfc5054#section-2.5.4, the value is 4 times smaller than modulus (256 bits against 1024).
Using the wolfssl default behavior will drastically slow down the generation of shared secret (something like 10 times longer).
I think the API must not generate private key, or generate a more standard-sized one, as it is done in the patch below : 

```
---
 wolfcrypt/src/srp.c     | 2 +-
 wolfssl/wolfcrypt/srp.h | 3 +++
 2 files changed, 4 insertions(+), 1 deletion(-)

diff --git a/wolfcrypt/src/srp.c b/wolfcrypt/src/srp.c
index 73abb6a..3bbfe8f 100644
--- a/wolfcrypt/src/srp.c
+++ b/wolfcrypt/src/srp.c
@@ -512,7 +512,7 @@ int wc_SrpGetPublic(Srp* srp, byte* pub, word32* size)
 
     /* priv = random() */
     if (mp_iszero(&srp->priv) == MP_YES)
-        r = wc_SrpGenPrivate(srp, pub, modulusSz);
+        r = wc_SrpGenPrivate(srp, pub, SRP_PRIVK_MIN_BITS);
 
     /* client side: A = g ^ a % N */
     if (srp->side == SRP_CLIENT_SIDE) {
diff --git a/wolfssl/wolfcrypt/srp.h b/wolfssl/wolfcrypt/srp.h
index f09432c..95b7bd3 100644
--- a/wolfssl/wolfcrypt/srp.h
+++ b/wolfssl/wolfcrypt/srp.h
@@ -51,6 +51,9 @@
 /* Set the minimum number of bits acceptable in an SRP modulus */
 #define SRP_DEFAULT_MIN_BITS 512
 
+/* Set the minimum number of bits acceptable for private keys (RFC 5054)*/
+#define SRP_PRIVK_MIN_BITS 256
+
 /**
  * SRP side, client or server.
  */
-- 

```

kaleb-himes, 2016-11-03:
Hi damv,

Thank you for contacting wolfSSL in regards to this issue. I will bring up your suggestion at our next team meeting and get back to you as soon as possible.

Kind Regards,

Kaleb and the wolfSSL Team

kaleb-himes, 2016-11-11:
Hi damv,

One of our engineers is in the process of evaluating this. We should have another update for you soon.

Kind Regards,

Kaleb & the wolfSSL Team

moisesguimaraes, 2016-11-21:
Hi @damv,

Thanks for your suggestion, a fix for this issue is in its way to our master branch.

kaleb-himes, 2016-11-21:
This has been addressed in PR: #639

closed 2016-11-21 by moisesguimaraes