wolfSSL/wolfssl #612: v3.9.8-29-g17a34c5 breaks retransmissions in DTLS server

jonasconnode, 2016-11-03:
WolfSSL in DTLS server mode no longer handles repeated client_hello messages from a client — they are just dropped. Expected behaviour is to handle the repeated client_hello and resend a hello_verify_request. This can happen if the first hello_verify_request from the server gets dropped and the client retransmits a client_hello.

I believe this was broken by this change in v3.9.8-29-g17a34c5:

```
@@ -7807,10 +7807,13 @@ static int DoDtlsHandShakeMsg(WOLFSSL* ssl, byte* input, word32* inOutIdx,
     }
     else {
         /* This branch is in order next, and a complete message. */
-        ssl->keys.dtls_expected_peer_handshake_number++;
         ret = DoHandShakeMsgType(ssl, input, inOutIdx, type, size, totalSz);
-        if (ret == 0 && ssl->dtls_msg_list != NULL)
-            ret = DtlsMsgDrain(ssl);
+        if (ret == 0) {
+            ssl->keys.dtls_expected_peer_handshake_number++;
+            if (ssl->dtls_msg_list != NULL) {
+                ret = DtlsMsgDrain(ssl);
+            }
+        }
```

This change moves the incrementing of the expected handshake number to after the call to DoHandShakeMsgType(). The expected number is reset to zero in DoClientHello(), but now it gets incremented after that.

kaleb-himes, 2016-11-03:
Hi jonasconnode,

Thank you for contacting wolfSSL. I'll get in touch with our DTLS expert at our next meeting and get clarification for you on why that change was made.

Regards,

Kaleb and the wolfSSL Team

ejohnstown, 2016-11-03:
That commit was for the support for the asynchronous crypto support. I have a fix for this that I'm testing.

closed 2016-12-13 by ejohnstown