wolfSSL/wolfssl #627: SSL verify callback with peer chain information without the need for SESSION_CERTS

dr-duplo, 2016-11-11:
It would be nice to make the SSL peer chain information in the verify callback available without the need to enable SESSION_CERTS. It increases the ram usage of the SSL context dramatically, which is very critical especially for embedded devices with little resources.

My suggestion is to add new arguments to the verify callback (ssl.h:365) which point to the stack variable "certs" in DoCertificate() of internal.c:6290. With this addition one could do some processing with the chain without the need to store it.

Me and our project would appreciate the change.

kaleb-himes, 2016-11-11:
Hi dr-duplo,

Thank you for your suggestion. Our team will review and discuss in our next meeting. Once we have evaluated we will get back to you with our thoughts!

Kind Regards,

Kaleb & the wolfSSL Team

kaleb-himes, 2016-11-17:
Hi dr-duplo,

I sent an email to monsieur<dot>cm<at>gmx<dot>de if that is the incorrect email to use please send me a quick note kaleb<at>wolfssl<dot>com.

Thanks!

kaleb-himes, 2016-11-21:
Hi dr-duplo,

This item has been moved to zendesk support and assigned the case #2498. I am closing this github issue.

Regards,

Kaleb

PR #641 (2016-11-23): Add the peer cert buffer and count to X509_STORE_CTX for verify callback
Add the peer cert buffer and count to the X509_STORE_CTX used for the verify callback. Fixes #627.
commit 5b76a372 (2016-11-23): Add the peer cert buffer and count to the X509_STORE_CTX used for the verify callback. Fixes #627.
closed 2016-11-21 by kaleb-himes