wolfSSL/wolfssl #662: Documentation of SetIOSend contradicts itself.

rotschopf, 2016-12-13:
Hi,

in the pdf documentation page 288 the synopsis defines the callback as

`typedef int (*CallbackIOSend)(char *buf, int sz, void *ctx);`

while the Example given below the following function as callback:
```
/*Receive callback prototype*/
int MyEmbedSend(WOLFSSL *ssl, char *buf, int sz, void *ctx);
```

Judging from wolfssl/ssl.h:1120 the example is correct.

The same applies to SetIORecv.

ejohnstown, 2016-12-13:
The second version with the WOLFSSL object pointer is correct. I see the description for the ctx parameter is completely wrong, though. That is a pointer to your application defined I/O context data that you want your callback to have access to. It may be a reference to a data buffer to read from, or an object with buffers and all sorts of other state.

Thanks for the documentation bug report.

ejohnstown, 2016-12-14:
I have updated the manual master. Thanks again for the bug report.

closed 2016-12-14 by ejohnstown