wolfSSL/wolfssl #685: mingw minor compile problem for v3.10

jay, 2016-12-24:
I compiled wolfSSL 3.10 using mingw this evening and had one warning because I don't have SCTP enabled:
~~~
./wolfssl/test.h: In function 'tcp_socket':
./wolfssl/test.h:665:62: error: unused parameter 'sctp' [-Werror=unused-parameter]
cc1.exe: all warnings being treated as errors
~~~
In mingw wolfSSL configure treats most warnings as errors due to -Werror which made that a blocker so I did this:
~~~diff
diff --git a/wolfssl/test.h b/wolfssl/test.h
index e0a3c1a..330ca26 100644
--- a/wolfssl/test.h
+++ b/wolfssl/test.h
@@ -664,6 +664,9 @@ static INLINE void build_addr(SOCKADDR_IN_T* addr, const char* peer,
 
 static INLINE void tcp_socket(SOCKET_T* sockfd, int udp, int sctp)
 {
+#ifndef WOLFSSL_SCTP
+    (void)sctp;
+#endif
     if (udp)
         *sockfd = socket(AF_INET_V, SOCK_DGRAM, IPPROTO_UDP);
 #ifdef WOLFSSL_SCTP
~~~

And when I ran the tests I had one fail due to unit.test
~~~
starting unit tests...
 Begin API Tests
   wolfSSL_Init(): passed
   wolfCrypt_Init(): passed
   wolfSSL_CTX_use_certificate_buffer(): passed
SUCCESS4
wolfSSL error: tcp connect failed
FAIL: tests/unit.test.exe
~~~
In [test_client_nofail](https://github.com/wolfSSL/wolfssl/blob/v3.10.0-stable/tests/api.c#L807) it is trying to make the tcp client connect to 127.0.0.1 port 0 for some reason. i reported it earlier this year along with a bunch of other things that were fixed, this may have slipped through or be a regression, ref #356 @dgarske

JacobBarthelmeh, 2016-12-28:
Hi Jay,
Thanks for the report, am looking into this.
Regards,
Jacob

JacobBarthelmeh, 2017-01-03:
Pull Request #691 should resolve this issue.

jay, 2017-01-03:
Thanks Jacob. It resolves the issue I reported though it has uncovered another issue in unit.test, this one concerning crlf translation.

unit.test in [suites.c opens a tests/test.conf in text translation (not binary)](https://github.com/wolfSSL/wolfssl/blob/v3.10.0-stable/tests/suites.c#L364-L372), gets the actual size of the file and then attempts to read that size in. That fails because fread translates the CRLFs in test.conf into LFs.
fread attempts to read 36706 bytes but 34508 is returned causing an error

~~~
starting default cipher suite tests
notice: using default file tests/test.conf
read error
error from script 1
FAIL: tests/unit.test.exe
~~~

JacobBarthelmeh, 2017-01-03:
Hi Jay, that's great that it resolves the initial report. To replicate the new issue being seen what are you using with MinGW? Is it with MSYS, straight from Windows terminal or some IDE? Which version of Windows?

jay, 2017-01-03:
Windows 7 Enterprise x64, I'm using MSYS MINGW32 shell (bash) and I build wolfSSL using the autotools way ./configure foo ... && make. Also I'm using the original mingw and not mingw-w64.

JacobBarthelmeh, 2017-01-10:
Thanks Jay for the environment information. I was unable to reproduce the issue with fopen, tried mix matching the end of lines for some of the test files but could not get it to fail on my VM for testing. I do not doubt that the behavior exists and while continuing to try and replicate it I could see no issues with opening the test files in binary mode so added PR 706 to review.

jay, 2017-01-13:
Thanks Jacob. I built from master (v3.10.0-stable-105-g00fed61) this evening and now all the tests pass.

closed 2017-01-13 by jay