wolfSSL/wolfssl #720: SSL connection to google.com fails

rofl0r, 2017-01-19:
hi,

```sh
git clone git://github.com/rofl0r/rocksock
cd rocksock
./configure --with-ssl=wolfssl
make -j9
examples/http_test.out 172.217.6.206 443s
examples/http_test.c:53 - ssl error: revcd alert fatal error from rocksock_cyassl.c:72
```

fails only with google.com. works with github.com. works also with `examples/http_test.out irc.freenode.net 6697s`.
when built against libressl instead of wolfssl (3.10.0), it works also with google.com.
i suspect that google's custom server use some feature that wolfssl doesn't like.
i just happened to notice it when testing my library, so i thought i would let you know.

kaleb-himes, 2017-01-19:
Hi rofl0r,

We are aware of a requirement when connecting to google servers that has changed in the last 6-9 months. Google servers will outright ignore any connection attempt if the client hello packets does not include the supported curves extension.

Could you try adding these defines to your default settings when building the wolfSSL library with the intent of connecting to ANY google servers:

```
#define HAVE_SUPPORTED
#define HAVE_TLS_EXTENSIONS
```

(Corresponding configure option would be: "--enable-supportedcurves" which will define both of the above pre-processor macros)

and load at least one supported curve in any client application with the API:

```
wolfSSL_UseSupportedCurve(ssl, <CURVE NAME>);
```

A list of supported curves can be found in the comment section at the top of <wolfssl-root>/wolfcrypt/src/ecc.c

kaleb-himes, 2017-01-19:
Updated type in answer: HAVE_SUPPORTED should have been HAVE_SUPPORTED_CURVES

closed 2017-04-06 by kaleb-himes

rofl0r, 2017-04-07:
thanks for your reply, i didn't have time to further investigate this. since you closed it, does it mean that wolfssl now has the google compat stuff enabled by default ?

kaleb-himes, 2017-04-07:
Hi rofl0r,

That functionality has always been enabled by default, this is an application level requirement. The application needs to enable one or more supported curves. The action we took here (I apologize for not including when I closed this) was to add a section in our example client that shows how to do this:

https://github.com/wolfSSL/wolfssl/blob/master/examples/client/client.c#L1751

By changing that to "#if 1" you can turn on every supported curve that is on in the default build.

Best Regards,

Kaleb

rofl0r, 2017-04-07:
> This would be something that should be done in "rocksock" as well.

ok, will do.

> We are not familiar with "rocksock" or what it might be

since it's only one mouseclick away ( https://github.com/rofl0r/rocksock ) and has a README which explains what it is, i guess the above wording should better be `We are not interested in "rocksock" or what it is`.

anyway, thanks for your help!