wolfSSL/wolfssl #736: Inconsistent behaviour of wc_AesCtrEncrypt() implementations

jonasconnode, 2017-02-02:
The different implementations of AES-CTR mode have different requirement on the size argument:
- The software implementation will accept data of any size (storing non-block-sized data for the next call), which seems to be true also for PIC32 and Freescale
- The STM32F2_CRYPTO implementation must be called with a multiple of the block size, or it reads/writes beyond the buffers
- The WOLFSSL_STM32_CUBEMX implementation appears to only work on one block, so it must be called with exactly one AES block.

The documentation does not specify any constraints on the size argument.

jonasconnode, 2017-02-02:
Hmmm, actually the STM32 version of wc_AesCtrEncrypt() doesn't even implement AES-CTR properly. It doesn't keep track of the counter over multiple calls, instead it has a code line that looks like it's copied from the CBC code.

It's better to just use the "regular" platform independent version of wc_AesCtrEncrypt() on STM32 and let that call the STM32 version of wc_AesEncrypt(). That works.

dgarske, 2017-06-07:
Hi Jonas,

We resolved these STM32 issues in PR #807, which was merged into master on May 22, 2017. https://github.com/wolfSSL/wolfssl/pull/807

Here is the section of code from the commit which resolves AES CTR.
https://github.com/wolfSSL/wolfssl/commit/a8f08d1d704b4872880745f7e721a97bece26d96#diff-963f55ef9b69c3b01d1cd6976ddd4937R2988

Specifically this commit resolves and adds AES support on the STM32 F2/F4 hardware for AES Direct (ECB), CCM, CTR and GCM and supports non-block aligned input sizes.

Let us know if you have any other questions or issues.

Thanks,
David Garske, wolfSSL

closed 2017-06-07 by dgarske