wolfSSL/wolfssl #741: Minor documentation bugs for MD5 support in wolfcrypt

embray, 2017-02-07:
All the documentation on https://www.wolfssl.com/wolfSSL/Docs-wolfssl-manual-18-29-wolfcrypt-api-md5.html references a type named `md5`, but actually it should be `Md5`.

Also, `wc_InitMd5 `, `wc_Md5Update`, and `wc_Md5Final` are documented as returning `int`, when in fact they return `void`.

kaleb-himes, 2017-02-10:
Hi embray,

We have been updating the functions to return integer values instead of void so the return types can be checked. The developers doing that work updated the documentation as they should but some of the code is still waiting on code reviews since we are in the middle of a release cycle at the moment. The code should catch up with the documentation shortly!

I will take a look at the md5 references you pointed out and fix any discrepancies. Thanks for bringing this to our attention.

embray, 2017-02-12:
I see, in that case it *should* be up to date soon enough.

kaleb-himes, 2017-04-06:
waiting on PR #640

kaleb-himes, 2018-04-30:
Functionality added in PR #640

closed 2018-04-30 by kaleb-himes