wolfSSL/wolfssl #749: Build error, no definition of

apullin, 2017-02-10:
I am unable to build the WolfSSL library in the Arudino "IDE". I am pretty sure I followed exactly the steps: unpack, edit `wolfssl\wolfcrypt\settings.h` and uncomment `#define WOLFSSL_ARDUINO`, change dir to  `wolfssl/IDE/ARDUINO`, run `./wolfssl-arduino.sh `.

I am building with Arduino 1.8.1, using a bare (just setup() and loop(), empty) main program, with the WolfSSL added in.

Build output:
`"/Users/xyz/Library/Arduino15/packages/arduino/tools/arm-none-eabi-gcc/4.8.3-2014q1/bin/arm-none-eabi-gcc" -c -g -Os -w -std=gnu11 -ffunction-sections -fdata-sections -nostdlib --param max-inline-insns-single=500 -Dprintf=iprintf -MMD -mcpu=cortex-m3 -mthumb -DF_CPU=84000000L -DARDUINO=10801 -DARDUINO_SAM_DUE -DARDUINO_ARCH_SAM  -D__SAM3X8E__ -mthumb -DUSB_VID=0x2341 -DUSB_PID=0x003e -DUSBCON '-DUSB_MANUFACTURER="Arduino LLC"' '-DUSB_PRODUCT="Arduino Due"' "-I/Users/xyz/Library/Arduino15/packages/arduino/hardware/sam/1.6.11/system/libsam" "-I/Users/xyz/Library/Arduino15/packages/arduino/hardware/sam/1.6.11/system/CMSIS/CMSIS/Include/" "-I/Users/xyz/Library/Arduino15/packages/arduino/hardware/sam/1.6.11/system/CMSIS/Device/ATMEL/" "-I/Users/xyz/Library/Arduino15/packages/arduino/hardware/sam/1.6.11/cores/arduino" "-I/Users/xyz/Library/Arduino15/packages/arduino/hardware/sam/1.6.11/variants/arduino_due_x" "-I/Users/xyz/Documents/Arduino/libraries/wolfssl" "/Users/xyz/Documents/Arduino/libraries/wolfssl/bio.c" -o "/var/folders/dy/nsmz0p8s06b0z0_rrfvqtqz40000gn/T/arduino_build_274316/libraries/wolfssl/bio.c.o"
/Users/xyz/Documents/Arduino/libraries/wolfssl/bio.c:23:13: error: expected '=', ',', ';', 'asm' or '__attribute__' before 'long'
 WOLFSSL_API long wolfSSL_BIO_ctrl(WOLFSSL_BIO *bio, int cmd, long larg, void *parg)
             ^
/Users/xyz/Documents/Arduino/libraries/wolfssl/bio.c:36:1: error: unknown type name 'size_t'
 size_t wolfSSL_BIO_ctrl_pending(WOLFSSL_BIO *bio)
 ^
/Users/xyz/Documents/Arduino/libraries/wolfssl/bio.c:36:33: error: unknown type name 'WOLFSSL_BIO'
 size_t wolfSSL_BIO_ctrl_pending(WOLFSSL_BIO *bio)
                                 ^
/Users/xyz/Documents/Arduino/libraries/wolfssl/bio.c:69:30: error: unknown type name 'WOLFSSL_BIO'
 long wolfSSL_BIO_get_mem_ptr(WOLFSSL_BIO *bio, WOLFSSL_BUF_MEM **ptr)
                              ^
/Users/xyz/Documents/Arduino/libraries/wolfssl/bio.c:69:48: error: unknown type name 'WOLFSSL_BUF_MEM'
 long wolfSSL_BIO_get_mem_ptr(WOLFSSL_BIO *bio, WOLFSSL_BUF_MEM **ptr)
                                                ^
/Users/xyz/Documents/Arduino/libraries/wolfssl/bio.c:82:13: error: expected '=', ',', ';', 'asm' or '__attribute__' before 'long'
 WOLFSSL_API long wolfSSL_BIO_int_ctrl(WOLFSSL_BIO *bp, int cmd, long larg, int iarg)
             ^
/Users/xyz/Documents/Arduino/libraries/wolfssl/bio.c:93:36: error: unknown type name 'WOLFSSL_BIO'
 int wolfSSL_BIO_set_write_buf_size(WOLFSSL_BIO *bio, long size)
                                    ^
/Users/xyz/Documents/Arduino/libraries/wolfssl/bio.c:133:31: error: unknown type name 'WOLFSSL_BIO'
 int wolfSSL_BIO_make_bio_pair(WOLFSSL_BIO *b1, WOLFSSL_BIO *b2)
                               ^
/Users/xyz/Documents/Arduino/libraries/wolfssl/bio.c:133:48: error: unknown type name 'WOLFSSL_BIO'
 int wolfSSL_BIO_make_bio_pair(WOLFSSL_BIO *b1, WOLFSSL_BIO *b2)
                                                ^
/Users/xyz/Documents/Arduino/libraries/wolfssl/bio.c:167:41: error: unknown type name 'WOLFSSL_BIO'
 int wolfSSL_BIO_ctrl_reset_read_request(WOLFSSL_BIO *b)
                                         ^
/Users/xyz/Documents/Arduino/libraries/wolfssl/bio.c:182:24: error: unknown type name 'WOLFSSL_BIO'
 int wolfSSL_BIO_nread0(WOLFSSL_BIO *bio, char **buf)
                        ^
/Users/xyz/Documents/Arduino/libraries/wolfssl/bio.c:210:23: error: unknown type name 'WOLFSSL_BIO'
 int wolfSSL_BIO_nread(WOLFSSL_BIO *bio, char **buf, int num)
                       ^
/Users/xyz/Documents/Arduino/libraries/wolfssl/bio.c:258:24: error: unknown type name 'WOLFSSL_BIO'
 int wolfSSL_BIO_nwrite(WOLFSSL_BIO *bio, char **buf, int num)
                        ^
/Users/xyz/Documents/Arduino/libraries/wolfssl/bio.c:325:23: error: unknown type name 'WOLFSSL_BIO'
 int wolfSSL_BIO_reset(WOLFSSL_BIO *bio)
                       ^
/Users/xyz/Documents/Arduino/libraries/wolfssl/bio.c:355:25: error: unknown type name 'WOLFSSL_BIO'
 long wolfSSL_BIO_set_fp(WOLFSSL_BIO *bio, XFILE fp, int c)
                         ^
/Users/xyz/Documents/Arduino/libraries/wolfssl/bio.c:355:43: error: unknown type name 'XFILE'
 long wolfSSL_BIO_set_fp(WOLFSSL_BIO *bio, XFILE fp, int c)
                                           ^
/Users/xyz/Documents/Arduino/libraries/wolfssl/bio.c:375:25: error: unknown type name 'WOLFSSL_BIO'
 long wolfSSL_BIO_get_fp(WOLFSSL_BIO *bio, XFILE* fp)
                         ^
/Users/xyz/Documents/Arduino/libraries/wolfssl/bio.c:375:43: error: unknown type name 'XFILE'
 long wolfSSL_BIO_get_fp(WOLFSSL_BIO *bio, XFILE* fp)
                                           ^
/Users/xyz/Documents/Arduino/libraries/wolfssl/bio.c:393:32: error: unknown type name 'WOLFSSL_BIO'
 int wolfSSL_BIO_write_filename(WOLFSSL_BIO *bio, char *name)
                                ^
/Users/xyz/Documents/Arduino/libraries/wolfssl/bio.c:419:22: error: unknown type name 'WOLFSSL_BIO'
 int wolfSSL_BIO_seek(WOLFSSL_BIO *bio, int ofs)
                      ^
/Users/xyz/Documents/Arduino/libraries/wolfssl/bio.c:437:37: error: unknown type name 'WOLFSSL_BIO'
 long wolfSSL_BIO_set_mem_eof_return(WOLFSSL_BIO *bio, int v)
                                     ^
Using library wolfssl in folder: /Users/xyz/Documents/Arduino/libraries/wolfssl (legacy)
exit status 1
Error compiling for board Arduino Due (Programming Port).`

kaleb-himes, 2017-02-23:
Hi Apullin,

Thanks for opening this issue we suspect it is just due to a recently added new file in wolfSSL. I'll take a look at the arduino script and make any necessary adjustments.

Regards,

Kaleb

pasko-zh, 2017-04-17:
@kaleb-himes, I have a similar Issue #859. So compiling only works for Version 3.9.0 of wolfssl.

closed 2018-04-30 by kaleb-himes