wolfSSL/wolfssl #751: segmentation fault issue

hajjihraf, 2017-02-13:
Hi, 
I encountered a segmentation fault issue in the API wc_Sha256Update.

attached are the "wolfServer.c", "wolfClient.c" and "user_settings.h" that I used in my testing .

[src.zip](https://github.com/wolfSSL/wolfssl/files/770754/src.zip)

Thanks,

JacobBarthelmeh, 2017-02-17:
Thank you hajjiharaf for providing the example and bringing this to our attention. We will review the issue, and provide feedback.

Regards,
Jacob

JacobBarthelmeh, 2017-03-01:
hajjiharaf,

As an update on this I was able to duplicate the behavior seen and have found the root cause. Pull request #771 was opened to fix it.

Regards,
Jacob

hajjihraf, 2017-03-01:
Hi Jacob,

Thanks for the update :)

Best Regards,
Achraf

closed 2017-03-13 by JacobBarthelmeh